espressif/esptool #425: Connection Failure on Windows: "Timed out waiting for packet header"

StevenLawsonBAF, 2019-04-18:
* Operating system: Windows 7
* Python version: Python 3.7.3
* ESP hardware in use: ESP32-WROOM-32D

# Full esptool.py command line as run: 
```
esptool.py -p COM4 --chip=esp32 read_mac
```
# Full output from esptool.py:
```
esptool.py v2.6
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```
# What is the expected behaviour?
ESPTool should connect to the ESP32 device and retrieve MAC address.
# Do you have any other information from investigating this?

- **I can use the same hardware, connections, and computer on a Linux OS, and ESPTool works fine, every time.**
- This is the ```--trace``` output from Windows: https://drive.google.com/open?id=134nem08Gbyvfq-PUiLIC5K-53F3ZAxSM
- I am using an Adafruit CP2104 breakout as a USB to serial converter. I've uninstalled/reinstalled the Windows drivers for this.
- I've also uninstalled/reinstalled Python and ESPTool.
- My logic analyzer shows that the EN & IO0 signal for entering the bootloader toggles the same in both Windows and Linux:

**Windows:**
[image]
[Saleae Logic Capture File](https://drive.google.com/file/d/1wYpUTCabX53ejqln2GDoKjF3eAirGv06/view?usp=sharing)

**Linux:**
[image]
[Saleae Logic Capture File](https://drive.google.com/file/d/1dsUNBEBExAaMTrFmeQPTKHJX_IdVZTDo/view?usp=sharing)

projectgus, 2019-04-23:
Hi @StevenLawsonBAF,

Thanks for the detailed report.

> My logic analyzer shows that the EN & IO0 signal for entering the bootloader toggles the same in both Windows and Linux:

This is unfortunately not what the capture shows. On Windows there is a couple of milliseconds delay after EN goes high and before IO0 goes low, and the ROM has started sending serial output before IO0 goes low (meaning the chip has already "bootstrapped" with IO0 high):

[image]

On Linux the transitions happen at almost the exact same time:

[image]

This is mostly a limitation of Windows serial drivers and it's hard to work around.

It looks like the hardware board you are using has the "auto reset circuit", is that right? If so, add a capacitor between the EN and GND pins - anything in the range 100nF or higher should be enough. Most good quality development boards have this already, but some vendors still leave it off.

More details here: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#automatic-bootloader

projectgus, 2019-05-28:
Hi @StevenLawsonBAF ,

Did you get a chance to look at the reply above and try the capacitor mentioned?

If you have any other details about the auto-reset circuit in use, feel free to post them.

StevenLawsonBAF, 2019-05-28:
@projectgus - thanks for your reply, and my apologies for not getting back to you sooner.

I did try adding a capacitor (as you mentioned), however it didn't solve the problem. That being said, I didn't spend much time troubleshooting with a logic analyzer or trying different capacitor values because I was able to bypass the issue by switching from the CP2104 to a FTDI-based USB->serial module, which seems to be working as expected.

I've attached the relevant bits of the schematic for the specific device I'm working with.

[image]

projectgus, 2019-05-28:
Thanks @StevenLawsonBAF . Did you add the capacitor between WIFI_EN & GND? I think because EN is driven with the AND gate in your design, it would need to be there to make any difference.

FTDI probably works because the Windows driver is setting the pins at roughly the same time, compared to the CP2102 driver.

I don't think there's anything we can do to fix this on the esptool.py side, unfortunately.

projectgus, 2019-10-18:
Closing as I don't think we can do anything from the esptool.py side.

closed 2019-10-18 by projectgus