espressif/esptool #426: 乐鑫官网买的2个ESP32-DevKitC, 一个esp-wroom-32 ，一个esp-wroom-32D, USB 烧录都正常，自己接线烧录32是正常，32D 老是 timeout waiting for packet header

philar, 2019-04-21:
问题如题描述，接线方式32 和32D 都是一样的，电源，rxtx，io0 拉低，网上的各种方式：
1.出现connecting的时候，长按boot
2.出现connecting的时候，长按EN 
3.io2 也拉低
4.换成115200的，比较低的波特率
5.用5v的电压
以上的方式都试过，32D 始终烧录不了，32 就很接好线就能烧录，什么都不用做
有没有遇到类似问题的兄弟，希望能提供一些建议，或者方向

projectgus, 2019-04-23:
> 2.出现connecting的时候，长按EN

如果您长按boot但是不按EN, 可以烧录吗? 如果烧录不了, esptool.py 有什么输出?

philar, 2019-04-23:
长按boot，不按EN，也是不可以烧录的，esptool一开始就是___……___……___……，是很多次之后就是输出错误信息，timeout waiting for packet header

发自我的 iPhone

reatalio, 2019-05-20:
Perdonar por escribir en mi idioma, espero que el traductor le pueda ser de ayuda, tuve el mismo problema en varias placas goouuu-esp32, la unica forma de poder realizar la carga era presionar boot (ioO) y luego conectar la alimentación usb( sin soltar el pulsador boot) y proceder a la descarga del programa desde arduino IDE, esto puede ser debido a un cortocircuito en la resistencia pull-up del EN en el ESP32,  entre los pines 3.3v y EN, comprueba entre los pines EN y 3.3v que existen aproximadamente 11-20kohms y tambien comprobar pulsador boot(ioO) manteniendo pulsado, entre GND y EN que debe haber 470ohms. Si esas medidas son erroneas es posible que el autoboot no se pueda efectuar.
Espero que pueda resolver su problema

boys-no-time, 2019-06-25:
我也遇到了这个问题，初步怀疑和这一段描述有关：
esptool.py can automatically enter the bootloader on many boards by using the RTS and DTR modem status lines to toggle GPIO0 and EN automatically.（参见快速入门-与 ESP32 创建串口连接的注解）
按上面的描述ESP32-DevKitC开发板烧录程序的时候应该是不需要操作boot0或EN的，自动会烧录。但实际情况是有时候可以有时候又不行，提示connect超时。还是需要按boot0才能烧录程序。而我们自己采购模组开发的产品是直接将PIN34 35的串口直接拉出来，然后在外部使用串口调试工具进行烧录，对这个DTR和RTS并没有和EN IO0进行连接，所以即使按下boot0也还是提示timeout waiting for packet header #426 甚至上电能打开串口也看不到调试信息。所以现在有两个问题：
1.即使参照ESP32-DevKitC-V4_SCH_20180607A.pdf原理图中的开发板为什么有时可以有时又不可以自动烧录程序；
2.如果我们自己采购模组开发产品，将PIN34 35的串口直接拉出来，然后在外部使用串口调试工具进行连接，这样操作上电能打开串口却看不到任何调试信息（偶尔有段时间有调试信息输出）而且也不能烧录程序。那么我们自己采购模组开发产品，如何处理串口接线才是合适的？

radimkarnis, 2021-03-22:
I am closing this issue because the description points to a HW issue and it seems like it is not relevant anymore. Please reopen if you think further discussion is needed.

closed 2021-03-22 by radimkarnis