espressif/esptool #427: Esptool.py: Command not found

degofedal, 2019-04-23:
Hi,

I'm having trouble getting esptool up and running on my system. I seem to have all the required assets installed but I'm getting a "command not found" when trying to run "esptool.py" in the terminal.

I'm preparing to flash a couple of Shelly 1's but have no clue where to look for a solution.

Any help will be greatly appreciated. Thanks in advance :) 

* Operating system: macOS 14.4.4
* Python version: 2.7

# Full esptool.py command line as run:
`Last login: Tue Apr 23 13:49:08 on ttys000
Mac-Mini:~ jeppeladefoged$ esptool.py
-bash: esptool.py: command not found
Mac-Mini:~ jeppeladefoged$ `

Esptool installed:
[image]

Python version:
[image]

Command not found error:
[image]

Pyserial installed:
[image]

projectgus, 2019-04-24:
Hi @degofedal,

Thanks for the detailed description of the problem. Most likely the "bin" directory where esptool.py has been installed is not on your PATH.

If you run `pip show -f esptool` then it should list all of the files installed by the esptool package. One of these files will look like `something/bin/esptool.py` (the `/bin`/ is the important part).

You can either run esptool.py directly by providing the full path to it on the command line, or you can [edit the PATH environment variable](https://thecustomizewindows.com/2013/11/mac-os-x-change-path-variable-guide/) so this directory is included on the PATH and you can type the name `esptool.py` to run the tool.

(Note that the path printed by pip may be a relative path, so you'll need to convert it to an absolute path if adding to the PATH variable.)

degofedal, 2019-04-24:
Hi @projectgus 

Thank you so much for your detailed guidance! That fixed my problem and i managed to find the absolute path `/Users/USERNAME/Library/Python/2.7/bin/esptool.py` to esptool.py via the command `pip show -f esptool` you supplied. Editing the PATH environment seems above my pay grade for now.

Now all I have to do I wait for the Shelly 1's to arrive :) 

Once again, thank you! Your help is greatly appreciated.

projectgus, 2019-04-24:
No problems, glad it's working. If you've got esptool.py running then I'll close this issue.

closed 2019-04-24 by projectgus

IamSandeepKrishnan, 2019-12-11:
thank you I was looking for the same issue resolution

xiaogong331, 2022-01-13:
Thinks for your issue, I meet the same problem