espressif/esptool #429: Invalid head packet of... and timed out waiting for packet header all over the place.

DexterHill0, 2019-05-08:
I recently had a problem where my Mac decided to no longer recognise the esp32. (it came up with the 'an accessory is using too much power..' warning) I had to re-install the drivers to let it recognise the esp32 again. Afte thsi incident, however, I have been getting 'Invalid head packet of' and 'timed out waiting for packet header' when uploading my scripts. 9/10 times I get one of these two errors. I can't seem to find any thing that stops it from uploading (e.g. extra devices like light strips) it just kind of 'happens'. At the end of my 'Invalid head packet of' I have had:
(0x30)
(0x31)
(0x69)
(0x0A)
(0x0D)
(0x52)?
and definately others I am missing.

Has the incident damaged my esp32 and that is why it isn't properly uploading, or is there a fix to this mess?

Thanks,
D

(My arduino output)
[image]

projectgus, 2019-05-09:
I don't think your ESP32 is damaged, but it seems like something is corrupting the serial connection. esptool is expecting to see the byte 0x0C to start the packet and instead it's receiving what looks like corrupt bytes, or could be it's booting into the wrong boot mode.

Exactly what hardware (board, etc) do you have? Is anything connected to the ESP32 board itself?

If your board has a "Boot" button can you hold it down when flashing? Any difference?

DexterHill0, 2019-05-09:
Thanks for the reply, I bought this board: [https://www.amazon.co.uk/ESP-WROOM-32-Development-Dual-Mode-Microcontroller-Integrated/dp/B0718T232Z/ref=sr_1_6?keywords=esp32&qid=1557414705&s=gateway&sr=8-6](url). Currently, the only things I have attached to the board are an sd card reader (that is not even being used) and some light strips, that I am using for testing.

My board doesn't have a button that clearly says boot on it, but it does have two buttons, one saying 'EN' and the other saying what looks like '100', which I believe is the equivalent of the boot button.

By holding down this button, it seems that everything it flashes _much_ faster. However, this visibly did nothing to help uploading of the sketch - it still errors.

projectgus, 2019-05-13:
The button probably says IO0, that's the same as Boot button.

Suggest checking the things which are connected (even if not used) against the pins mentioned here:
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

projectgus, 2019-05-28:
Hi @DexterHill0 ,

Did you make any progress debugging this?

Angus

DexterHill0, 2019-06-02:
Sorry for the long wait, I totally forgot about this thread.

I unplugged some devices and moved around some pins, and now it seems to upload. I also optimised the code so it is a bit shorter (wether that did anything, I don't know)

Thanks for the link, it really helped debugging.

projectgus, 2019-06-03:
Glad to hear it's working. Possibly something was connected to a strapping pin which should not have been.

I'm going to close this issue if the problem is solved.

closed 2019-06-03 by projectgus