espressif/esptool #430: 我购买的lolin d32 传感器出现 A fatal error occurred: Timed out waiting for packet header

BLY970705, 2019-05-14:
(Please delete any lines which don't apply.)

* Operating system:win 10
* Python version: (`python -V` to check this)
* ESP hardware in use:
Lolin D32 Pro

我在测试APDS-9301时出现的这个问题，这个程序是库文件自带的example

int 引脚连接在2上

#include "Wire.h"
#include <Sparkfun_APDS9301_Library.h>

APDS9301 apds;

#define INT_PIN 2 // We'll connect the INT pin from our sensor to the
// INT0 interrupt pin on the Arduino.
bool lightIntHappened = false; // flag set in the interrupt to let the
//  mainline code know that an interrupt occurred.

void setup()
{
  delay(5);    // The CCS811 wants a brief delay after startup.
  Serial.begin(115200);
  Wire.begin();

  // APDS9301 sensor setup.
  apds.begin(0x39);  // We're assuming you haven't changed the I2C
  //  address from the default by soldering the
  //  jumper on the back of the board.
  apds.setGain(APDS9301::LOW_GAIN); // Set the gain to low. Strictly
  //  speaking, this isn't necessary, as the gain
  //  defaults to low.
  apds.setIntegrationTime(APDS9301::INT_TIME_13_7_MS); // Set the
  //  integration time to the shortest interval.
  //  Again, not strictly necessary, as this is
  //  the default.
  apds.setLowThreshold(0); // Sets the low threshold to 0, effectively
  //  disabling the low side interrupt.
  apds.setHighThreshold(50); // Sets the high threshold to 500. This
  //  is an arbitrary number I pulled out of thin
  //  air for purposes of the example. When the CH0
  //  reading exceeds this level, an interrupt will
  //  be issued on the INT pin.
  apds.setCyclesForInterrupt(1); // A single reading in the threshold
  //  range will cause an interrupt to trigger.
  apds.enableInterrupt(APDS9301::INT_ON); // Enable the interrupt.
  apds.clearIntFlag();

  // Interrupt setup
  pinMode(INT_PIN, INPUT_PULLUP); // This pin must be a pullup or have
  //  a pullup resistor on it as the interrupt is a
  //  negative going open-collector type output.
  attachInterrupt(digitalPinToInterrupt(INT_PIN), lightInt, FALLING);
  Serial.println(apds.getLowThreshold());
  Serial.println(apds.getHighThreshold());
}

void loop()
{
  static unsigned long outLoopTimer = 0;
  apds.clearIntFlag();

  // This is a once-per-second timer that calculates and prints off
  //  the current lux reading.
  if (millis() - outLoopTimer >= 1000)
  {
    outLoopTimer = millis();

    Serial.print("Luminous flux: ");
    Serial.println(apds.readCH0Level(), 6);

    if (lightIntHappened)
    {
      Serial.println("Interrupt");
      lightIntHappened = false;
    }
  }
}

void lightInt()
{
  lightIntHappened = true;
}

我曾经成功运行过这个程序一次，几天之后再次使用的时候先是无法读数，接着就显标题的错误了，这是什么原因引起的啊

Alvin1Zhang, 2019-05-14:
@BLY970705 在 ESP32 中，Pin 2 在复位时必须置低以进入 serial boot loader 模式。烦请参考 ESP32 Datasheet 中第 2.4 节 "Strapping Pins"。建议将 INT 连接至其他引脚。谢谢。

BLY970705, 2019-05-14:
> @BLY970705 在 ESP32 中，Pin 2 在复位时必须置低以进入 serial boot loader 模式。烦请参考 ESP32 Datasheet 中第 2.4 节 "Strapping Pins"。建议将 INT 连接至其他引脚。谢谢。

非常感谢您的回复，我第一次学习这方面的知识很抱歉问了这么初始的问题，我第一次学习这方面的知识OTZ

Alvin1Zhang, 2019-05-14:
@BLY970705 你的问题解决了吗？此问题是否可以关闭？谢谢。

closed 2019-05-14 by BLY970705