espressif/esptool #431: configure: error: missing required tool: gobjcopy objcopy

g30r93g, 2019-05-15:
* Operating system: MacOS
* Python version: 2.7.10 and 3.6.4
* ESP hardware in use: ESP-8266-12E (NodeMCU v1.0)

# Full esptool.py command line as run:

`make toolchain esptool libhal STANDALONE=n`

# Full output from esptool.py (please copy and paste all lines of output)

```
Makefile:186: warning: overriding commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
Makefile:181: warning: ignoring old commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
/Applications/Xcode-beta.app/Contents/Developer/usr/bin/make -C crosstool-NG -f ../Makefile _ct-ng
../Makefile:186: warning: overriding commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
../Makefile:181: warning: ignoring old commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
checking build system type... x86_64-apple-darwin18.6.0
checking host system type... x86_64-apple-darwin18.6.0
checking for a BSD-compatible install... /usr/local/bin/ginstall -c
checking for grep that handles long lines and -e... /usr/local/bin/ggrep
checking for egrep... /usr/local/bin/ggrep -E
checking for a sed that does not truncate output... /usr/local/bin/gsed
checking whether sed understands -r -i -e... yes
checking whether ln -s works... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for ranlib... ranlib
checking for gobjcopy... no
checking for objcopy... no
configure: error: missing required tool: gobjcopy objcopy
make[1]: *** [_ct-ng] Error 1
make: *** [crosstool-NG/ct-ng] Error 2
```

# What is the expected behaviour?

The code should compile.

# Do you have any other information from investigating this?

Trying to use esp-homekit-demo to create roller blinds.
https://github.com/maximkulkin/esp-homekit-demo
https://github.com/pfalcon/esp-open-sdk

projectgus, 2019-05-16:
Hi @g30r93g ,

esptool.py is a python tool for flashing ESP8266 & ESP32 chips. It doesn't have any build process that uses "make".

This error looks like it's coming when building esp-open-sdk, so please try reporting it there (or to whatever repository you're running "make" in, if not esp-on-sdk). Most likely the solution is to install a copy of the utility "objcopy" on the computer you're building on.

closed 2019-05-16 by projectgus

g30r93g, 2019-05-16:
Doh too many tabs and I got confused. Sorry for that!

g30r93g, 2019-05-16:
Thanks also for pointing me in the right direction!

projectgus, 2019-05-17:
@g30r93g No problems, happens to everyone sometimes. :)