espressif/esptool #432: ESP8266 on upload error

amttheking, 2019-05-22:
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM9
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\nielit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/nielit/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main
    esp.connect(args.before)
  File "C:/Users/nielit/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
_

* Operating system:
* Python version: (`python -V` to check this)
* ESP hardware in use:

# Full esptool.py command line as run:

# Full output from esptool.py (please copy and paste all lines of output)

# What is the expected behaviour?

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2019-05-23:
Hi @amttheking ,

It's not possible to provide any support from the limited information you've provided. Please edit your issue and fill out the exact information which is asked for in the issue template.

Please also go through the Troubleshooting list here and see if any of these steps help: https://github.com/espressif/esptool/#troubleshooting

amttheking, 2019-05-23:
Hi  Projectus,
i am using 64 bit windows8 machine. above mentioned Error is thrown many times whenever i am trying to upload any ESP Code.
Ask which kind of details you need to troubleshoot this error.

projectgus, 2019-05-23:
Please read the first post in the issue. There are a list of dot points to fill in. Please fill them in.

Most important detail, exactly what hardware are you using?

projectgus, 2019-05-28:
@amttheking Any progress with this? If not, can you provide those extra details please?

shrakanis, 2019-05-30:
same for me, using USB-TTL shanger and esp-12F

istiyakaminsanto, 2019-05-31:
Arduino: 1.8.6 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Sketch uses 279880 bytes (26%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28944 bytes (35%) of dynamic memory, leaving 52976 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\istiy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/istiy/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main
    esp.connect(args.before)
  File "C:/Users/istiy/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
_
the selected serial port _
 does not exist or your board is not connected

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

shrakanis, 2019-06-01:
returned to earlier version an it works again...

boards manage search for esp

esp8266

install 2.5.0

projectgus, 2019-06-03:
@shrakanis Are you saying esptool 2.5 works but 2.6 doesn't? Can you post the output from a working flash, please?

shrakanis, 2019-06-03:
@projectgus 

Sketch uses 351248 bytes (33%) of program storage space. Maximum is 1044464 bytes.
Global variables use 43004 bytes (52%) of dynamic memory, leaving 38916 bytes for local variables. Maximum is 81920 bytes.
C:\Users\shrakanis\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9/esptool.exe -vv -cd nodemcu -cb 115200 -cp COM4 -ca 0x00000 -cf C:\Users\SHRAKA~1\AppData\Local\Temp\arduino_build_846398/siustuvas_v4_temp_2.ino.bin 
esptool v0.4.13-1-gf80ae31 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM4
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 355392 bytes from C:\Users\SHRAKA~1\AppData\Local\Temp\arduino_build_846398/siustuvas_v4_temp_2.ino.bin to flash at 0x00000000
	erasing flash
	size: 056c40 address: 000000
	first_sector_index: 0
	total_sector_count: 87
	head_sector_count: 16
	adjusted_sector_count: 71
	erase_size: 047000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 15000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 22% ]
................................................................................ [ 45% ]
................................................................................ [ 68% ]
................................................................................ [ 91% ]
............................                                                     [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete

HenniC4, 2019-06-28:
I got the same behaviour when I was trying to program a couple of sonoff switches on Linux with tasmota, using the cheap red PCB fake FTDI type of converter that everyone has. Checked everything on the scope, connections all fine, Button connected to GPIO0, 3,3V nice and clean, signals good, it just didn't work, the answer from the esp8285 was stuck at logic 0. I then disconnected ES (TX) --> FTDI (RX) restarted esptool and saw correct UART signals from the Controller. Then (while esptools prints dots and lines) reconnected the wire and it got correctly programmed. This seems to be reproducible. Before, I failed on dozends of tries with two different Sonoffs, after I have programmed two more boards at the first try.

denisfou, 2019-07-14:
Same for me, esptools 2.6 seems to don't send correctly binaries to ESP (ESP01 and nodemcu have same problem) and don't reset the device. 
I have to downgrade package ESP community to 2.5.0 to make it work.

projectgus, 2019-07-15:
@denisfou Are you using Windows as well?

denisfou, 2019-07-15:
> @denisfou Are you using Windows as well?

Yes i do.

mohammadrezasalehi95, 2019-07-20:
Same Error for me too, I tried Windows an Linux (ubuntu) both but still  same Error
I also need 2 NodeMcu 8266 12E, and one of them work, and the other one goes to that Error
(i downgrade my esp 8266 in Board Manager, but the same error as a result)

sandious, 2019-08-02:
I'm using Ubuntu 16.04 LTS.  After this "esptool.py"-Error I had a look at 
$ ls -l  ~/.arduino15/packages/esp8266/tools/python/3.7.2-post1 
drwxrwxr-x 2 yu yul  4 Aug  2 10:51 ./
drwxrwxr-x 3 yu yu  3 Aug  2 10:51 ../
-rw-r--r-- 1 ju ju  0 Mär 20 17:50 placeholder_for_arduino
lrwxrwxrwx 1 ju ju 15 Aug  2 10:51 _python -> /usr/bin/python*_

I tried to delete the softlink to /usr/bin/python and set a new one to
ln -s /usr/bin/python3.6

Afterwards it worked. Therefore its a ugly python-version problem.

projectgus, 2019-08-05:
@sandious Thanks for this information. Did you really get the error "esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header" from esptool.py, or were you getting a different error?

Please report this Python version problem to the ESP8266 Arduino project so they can fix it, as this part is not from esptool.py itself or even developed by Espressif: https://github.com/esp8266/Arduino/

sandious, 2019-08-05:
@projectgus 
> @sandious Thanks for this information. Did you really get the error "esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header" from esptool.py, or were you getting a different error?
Yes of course. That was the reason why I digged into esptool.py as the as the root of trouble apparently.

> Please report this Python version problem to the ESP8266 Arduino project so they can fix it, as this part is not from esptool.py itself or even developed by Espressif: https://github.com/esp8266/Arduino/
There is nothing to fix because the default-link within a linux-system for /usr/bin/* is correct. The point is "alternatives" python versions. Originally Ubuntu 16.04 comes with python-version 2.7 as default. So it is up to the user himself to choose the newest python-release anyway.

projectgus, 2019-08-05:
@sandious 
> Yes of course. That was the reason why I digged into esptool.py as the troublemaker apparently.

OK. I need to ask, some people reply "me too" to threads because of the most loose connections so it's good to be sure.

> There is nothing to fix because the default-link within a linux-system for /usr/bin/* is correct. The point is "alternatives" python versions. Originally Ubuntu 16.04 comes with python-version 2.7 as default. So it is up to the user himself to choose the newest python-release anyway.

Ah I mistakenly thought `/usr/bin/python*` was inviting some kind of wildcard expansion by the Arduino IDE but if it's actually a symlink to `/usr/bin/python` then I agree.

This is maybe a difference between pyserial behaviour on Python 2 & Python 3, then.

Do you mind double-checking that if you run `python2.7 esptool.py -p PORT flash_id` then it fails to connect, but `python3.6 esptool.py -p PORT flash_id` will succeed?

Can you please also confirm you have pyserial version 3.0.1 installed for both Python 2 & 3 (the Ubuntu packages are `python-serial` and `python3-serial`)?

sandious, 2019-08-05:
@projectgus 
 > OK. I need to ask, some people reply "

No worries!

> Do you mind double-checking that if you run `python2.7 esptool.py -p PORT flash_id` then it fails to connect, but `python3.6 esptool.py -p PORT flash_id` will succeed?

Exactly. In order to verify I always do tribble-checking.  

> Can you please also confirm you have pyserial version 3.0.1 installed for both Python 2 & 3 (the Ubuntu packages are `python-serial` and `python3-serial`)?
> EDIT: Never mind this question, I see ESP8266 Arduino ships their own copy of the pyserial module.

By the way I prefer to setup clean distribution software first. Therefore I had created an python3.6-ENV with pyserial 3.4 installed. It follows that the call from the symlink uses the latest version of pyserial.

mpopgun, 2019-08-12:
I can confirm, from board manager, esp8266 v2.5.0 works, 2.5.1 and 2.5.2 have the same error as described as above. 

Thanks!!

edoiks, 2019-08-14:
I have the same observations. My guess is when I hit upload on 2.5.2 there is additional step checking the info from esp board, then it tries to upload sketch and fails. But it stop failing when I set Erase Flash = Only Sketch. But maybe it is a different problem.

projectgus, 2019-08-14:
@edoiks If changing to "Erase Flash = Only Sketch" fixes the problem, then the problem is that Arduino IDE is running esptool.py twice (once to erase and once to write), and the second time it's failing to reset some boards into download mode. This issue is tracked on the Arduino repo, here: https://github.com/esp8266/Arduino/issues/6296

arfa123, 2019-08-15:
why esptool using python? what's the use of python in embedded systems programming

JoeBoJoe, 2019-08-16:
I am experiencing the same issue however my "Erase Flash" is set to "Only Sketch" and the problem still persists. Kinda sucks as I'm new to using the Arduino and the ESP8266 and spent the last day trying to figure out what I was doing wrong. Glad I found this thread eventually...

Strangely enough one of my uploads out of the many I tried did manage to go through without issue. I haven't been able to replicate that though.

Downgrading to version 2.5.0 has solved the issue though.

projectgus, 2019-08-18:
@JoeBoJoe What hardware (ESP8266 development board model) and OS (Windows, MacOS, etc) do you have?

JoeBoJoe, 2019-08-19:
> @JoeBoJoe What hardware (ESP8266 development board model) and OS (Windows, MacOS, etc) do you have?
I'm using an ESP-12F with an Arduino Uno on Windows 10.

Shahrizanijan7, 2019-08-19:
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/Asus/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main
    esp.connect(args.before)
  File "C:/Users/Asus/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
_

How about this? Help me guys.

JoeBoJoe, 2019-08-19:
@Shahrizanijan7 Downgrade esp8266 from 2.5.2 to v2.5.0 in board manager and see if it works then.

Shahrizanijan7, 2019-08-19:
> @Shahrizanijan7 Downgrade esp8266 from 2.5.2 to v2.5.0 in board manager and see if it works then.

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

It become this hrmm

JoeBoJoe, 2019-08-19:
> > @Shahrizanijan7 Downgrade esp8266 from 2.5.2 to v2.5.0 in board manager and see if it works then.
> 
> warning: espcomm_sync failed
> error: espcomm_open failed
> error: espcomm_upload_mem failed
> error: espcomm_upload_mem failed
> 
> It become this hrmm

Have you put your eps8266 into program mode before trying to upload?

Shahrizanijan7, 2019-08-19:
i use espreso lite v2 under esp8266

f5soh, 2019-08-19:
With 2.5.0 did you try both resetmethod set to 'ck' and 'nodemcu' ?

consolacion, 2019-08-20:
> @Shahrizanijan7 Downgrade esp8266 from 2.5.2 to v2.5.0 in board manager and see if it works then.

Yes, that worked

smtechchauhan, 2019-10-02:
i am facing this problem when i  compile ................

Arduino: 1.8.10 (Windows 8), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:30:0,

                 from C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from C:\Users\Ersumit\AppData\Local\Temp\Rar$DI02.047\Firebase_HomeAutomation\Firebase_HomeAutomation.ino:4:

C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:109:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

     std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

In file included from C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22:0,

                 from C:\Users\Ersumit\AppData\Local\Temp\Rar$DI02.047\Firebase_HomeAutomation\Firebase_HomeAutomation.ino:4:

C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:86:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

Multiple libraries were found for "ESP8266HTTPClient.h"
 Used: C:\Users\Ersumit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPClient
Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\Ersumit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi
Multiple libraries were found for "FirebaseArduino.h"
 Used: C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\Ersumit\Documents\Arduino\libraries\ArduinoJson-6.x
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

projectgus, 2019-10-08:
@smtechchauhan This is a compiler error somewhere in your Arduino project or one of the libraries which is in use. It's not related to esptool.

instinxt, 2019-10-22:
> esptool.py v2.6
> 2.6
> esptool.py v2.6
> Serial port COM3
> Connecting........_****.....****_....._****.....****_....._****.....****_.....____Traceback (most recent call last):
> File "C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in
> esptool.main(fakeargs)
> File "C:/Users/Asus/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main
> esp.connect(args.before)
> File "C:/Users/Asus/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect
> raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
> esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
> _
> 
> How about this? Help me guys.
Similar error for me 
I am using nodemcu 1.0

jphein, 2019-11-10:
I had to downgrade from 2.6.0 to v2.5.0.

klrock, 2019-11-11:
Unable to flash WLED

I used Updater to flash WLED_0.8.6_ESP8266.bin to my Lolin V3 nodemcu.
Lolin NodeMCU Features
• Arduino-Like Hardware IO
• Code like Arduino, but interactively in Lua script
• Event-driven API for network applications, which facilitates
developers writing code
• Integrates GPIO, PWM, IIC, 1-Wire and ADC all in one board
• 10 GPIO, every GPIO can be PWM, I2C, 1-wire
• 4M Flash Memory
• Built-in WiFi Antenna

Arduino IDE (I am using version 1.8.10) with the ESP8266 libraries and the Arduino core version 2.5.0.
Installed the NeoPixelBus library by Makuna (2.5.1), FastLED, ESPAsyncWebServer, ESPAsyncTCP (AsyncTCP for ESP32), and IRRemoteESP8266

Board: "Generic ESP8266 Module"
Upload Speed: "115200"
CPU Frequency: "80 MHz"
Crystal Frequency: "26 MHz"
Flash Size: "4M (1M SPIFFS)“
Flash Mode: "DOUT (compatible)"
Flash Frequency: "40MHz"
Reset Method: "ck"
Debug port: "Disabled"
Debug Level: ”None"
lwlP Variant: "v2 Lower Memory"
VTables: "Flash"
Exceptions: "Disabled"
Builtin Led: "2"
Erase Flash: "Only Sketch"
Port: ”COM7"
Get Board Info
Programmer: "AVR ISP"

Using Updater OTA started install of WLED_0.8.6_ESP8266.bin after putting the Lolin into flash mode. After a very short time I got this:

Your connection was interrupted
A network change was detected.
ERR_NETWORK_CHANGED

WLED-AP is not seen but a hidden network shows in the Wifi list.

I have repeated this three times with the same results.  

Please advise fix.

projectgus, 2019-11-12:
@klrock esptool is just the program that flashes your ESP8266 over serial. It seems like it's flashing fine, need to ask about this on a forum or on the [Arduino ESP8266 gitter](https://gitter.im/esp8266/Arduino) or something.

projectgus, 2019-11-12:
@jphein Can you give any details about the problem you had that was solved by the downgrade? And what hardware you are using?

jphein, 2019-11-12:
> @jphein Can you give any details about the problem you had that was solved by the downgrade? And what hardware you are using?

Np! I updated my original post.

f5soh, 2019-11-12:
> I had to downgrade from 2.6.0 to v2.5.0.
> ESP Generic Board
> This USB to serial TTL chip: CH340G UART / Programmer and ESP ESP8266 ESP-01S
> https://amzn.to/2rxKy5D with Schematic https://drive.google.com/open?id=16NHxje6vbX3h_fYh9n8_OzwX_KkquZy9

With basic programmer without dtr, the Reset method "no_dtr, no_sync" should work with 2.6.0

jphein, 2019-11-12:
> With basic programmer without dtr, the Reset method "no_dtr, no_sync" should work with 2.6.0

I found an option under Tools --> Reset Method -- None. That actually works. I just have to unplug it and plug it back in everytime.

f5soh, 2019-11-12:
Tools --> Reset Method -- None will be for 2.5.0 ?

projectgus, 2019-11-12:
@jphein That's weird. If the schematic you posted matches your hardware, your programmer has no way to reset the board programatically - RTS & DTR control lines are not connected to anything. I would expect that any reset method should work, but before flashing you have to set the switch to "PROG" mode, then unplug the ESP-01 module (or the whole board), then wait a couple of seconds and plug it back in. Otherwise the module will be in the wrong boot mode, and there's nothing esptool can do about this as it has no connected reset line - even if esptool tries a reset method, it won't do anything so same outcome.

jphein, 2019-11-12:
@f5soh You're right! I moved the Arduino folder, and it was misreporting my board version. I was still using 2.5.0 even though it said 2.6.0. 
I reinstalled it to replicate initial error. Made sure I was using 2.6.0. IT defaulted to "dtr (aka nodemcu)" reset method. I got the same error. Posted below. 

Then I tried your Reset method "no_dtr, no_sync". It worked!!! 

@projectgus, Yes, I make sure to unplug it, and plug it back in before I attempt flashing. The toggle switch is set to "prog"
`Arduino: 1.8.10 (Linux), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+111 (191024), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

fatal: not a git repository (or any of the parent directories): .git
Executable segment sizes:
IROM   : 370640          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27072   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1252  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 2316  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25088 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 401280 bytes (41%) of program storage space. Maximum is 958448 bytes.
Global variables use 28656 bytes (34%) of dynamic memory, leaving 53264 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.7
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a4:cf:12:c9:15:48
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4014
Detected flash size: 1MB
Hard resetting via RTS pin...
esptool.py v2.7
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
  File "/home/jp/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "/home/jp/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/esptool/esptool.py", line 2841, in main
    esp.connect(args.before)
  File "/home/jp/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

projectgus, 2019-11-13:
@jphein ah, I understand from the log now. Arduino has a custom `upload.py` script that runs esptool twice. This means the second time esptool runs it tries to reset the board again. Your board doesn't support programmatic reset, so it gets stuck. The discussion here is relevant: https://github.com/esp8266/Arduino/issues/6755#issuecomment-552758945

If using `no_dtr, no_sync` works then suggest sticking with that.

cseatec, 2019-11-22:
> I have to downgrade package ESP community to 2.5.0 to make it work.

Doing this downgrade from 2.6.0 to 2.5.0 was the key to my success.  I'm using an eBay ESP-01 board with 1M of flash.  My 3.3V (250mA from special FTDI cable) and TX/RX signals were pristine on the scope, so I didn't believe any of the other web articles about building an external power supply. I did go in and make sure the Wi-Fi wasn't auto connecting on power-up so it wasn't unnecessarily taxing the 3.3V source.  (Mine had starting autoconnecting after I connected it to my local Wi-Fi as a test via AT+ commands).  Now with 2.5.0, I can download firmware from Arduino IDE without any issues.  Many thanks to [denisfou](https://github.com/denisfou) for his comment above!

projectgus, 2019-11-25:
[Arduino ESP8266 Core version 2.6.1](https://github.com/esp8266/arduino/releases) includes fixes for their esptool integration that should fix most of the issues described in here (no longer tries to run esptool.py twice).

Updating to latest Arduino ESP8266 version should resolve most of the issues described here.

cseatec, 2019-11-25:
I just upgraded to 2.6.1 per your suggestion, and it is working. Thank you.  

I also believe that previously I may have been suffering from a subtle hardware configuration issue: I eventually figured out that powering up the entire ESP8266 board (ESP-01 in my case) with GPIO0 grounded is NOT a reliable way to make sure the ESP8266EX enters upload mode.  Instead, I now use a different order of operations:  1) Power up, 2) make sure GPIO0 is grounded, and 3) reset the micro one more time (temporarily ground reset pin). Doing this reliably places that the ESP8266 is in upload mode.  It's easy to confirm since no more debug data gets transmitted out of the UART0 port when reset is released.  This procedural change made the uploads work reliably for me, and 2.6.1 is now working with it as well.  Thanks again.

projectgus, 2019-11-28:
Closing this issue as I think most posters here will find success if updating to Arduino IDE 2.6.1.

If anyone is still experiencing problems flashing and you think it's an esptool problem, please open a new issue with the details.

closed 2019-11-28 by projectgus

tjclement, 2019-11-28:
For anyone looking at this thread after having made their own ESP32-WROOM board: you get this error too if GPIO12 is pulled up during boot. It should be left floating, or pulled down. Otherwise it undervolts its flash storage, causing this error to occur during writing to flash.

banknc, 2020-01-14:
Arduino: 1.8.10 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\bankk\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:30:0,

                 from C:\Users\bankk\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from C:\Users\bankk\Documents\Arduino\firebase_1\firebase_1.ino:2:

C:\Users\bankk\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:109:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

     std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

In file included from C:\Users\bankk\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22:0,

                 from C:\Users\bankk\Documents\Arduino\firebase_1\firebase_1.ino:2:

C:\Users\bankk\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:86:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

Multiple libraries were found for "FirebaseArduino.h"
 Used: C:\Users\bankk\Documents\Arduino\libraries\firebase-arduino-master
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\bankk\Documents\Arduino\libraries\ArduinoJson
Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\bankk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi
Multiple libraries were found for "ESP8266HTTPClient.h"
 Used: C:\Users\bankk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPClient
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
Invalid library found in C:\Program Files (x86)\Arduino\libraries\libraries: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\libraries
Invalid library found in C:\Program Files (x86)\Arduino\libraries\libraries: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\libraries

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

technicguy, 2020-02-15:
same for me, while uploading code to nodemcu esp8266, BUT after removed wire from (RX, TX) works for me.

instalator, 2020-03-21:
esptool.py v2.8
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\instalator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1/tools/upload.py", line 69, in <module>
    esptool.main(cmdline)
  File "C:/Users/instalator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/instalator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

jorenvh1, 2020-03-21:
Having the same issue... I'm using macos Catalina. I tried downgrading the ESP community version but nothing is working for me.

If I use 2.6 I get a python error: 
```
Fatal Python error: initsite: Failed to import the site module
Traceback (most recent call last):
  File "/usr/local/bin/../../../Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site.py", line 73, in <module>
    import os
  File "/usr/local/bin/../../../Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/os.py", line 28, in <module>
    from os.path import (altsep, curdir, defpath, devnull, extsep, pardir, pathsep,
ModuleNotFoundError: No module named 'os.path'; 'os' is not a package
exit status 1
Error compiling for board Generic ESP8266 Module.
```
On 2.5 I get this error:
```
esptool.py v2.6
2.6
esptool.py v2.6
Serial port /dev/cu.usbmodem1412201
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/Users/jorenvh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "/Users/jorenvh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 2653, in main
    esp.connect(args.before)
  File "/Users/jorenvh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
```

jorenvh1, 2020-03-22:
To add to my latest comment. For some reason it works from time to time whenever I switch usb ports before uploading to the board...

jorenvh1, 2020-03-23:
I think I found the problem.
I'm not sure if this is common knowledge but for me uploading does only work when I:

1) make sure the ESP8266 is not powered
2) press upload
3) during upload, connect the power to the esp8266

cseatec, 2020-03-23:
> I think I found the problem.
> I'm not sure if this is common knowledge but for me uploading does only work when I:
> 
> 1. make sure the ESP8266 is not powered
> 2. press upload
> 3. during upload, connect the power to the esp8266

jorenvh1 - As I mentioned in my reply to this thread back in November at [this link](https://github.com/espressif/esptool/issues/432#issuecomment-558067499), I found that in order to reliably cause the ESP8266 to enter boot mode, I had to power it up normally first, then hold the boot signal high, then pull the reset signal low (momentarily), release reset, then lastly release BOOT.  In this way, DC power is and remains stable when this additional reset occurs, and the ESP8266 then reliably enters boot mode.  Trying to force it into boot mode all at once on a cold power up wasn't working for me because the power to the BOOT signal wasn't always going high fast enough before the ESP8266 came out of reset.  Hopefully, this makes sense to you.

guga330, 2020-03-30:
Sketch uses 291292 bytes (30%) of program storage space. Maximum is 958448 bytes.
Global variables use 28832 bytes (35%) of dynamic memory, leaving 53088 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port /dev/cu.wchusbserial1440
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/Users/gustavus/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/Users/gustavus/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/Users/gustavus/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

Anyone help? I'm using MacOS Catalina, I have tried to downgrade ESP8266 Board to 2.5, I have the ESP8266 12E, I have tried holding the Flash button, all of this and no sucess.
I know it is trying to read the ESP8266 because when I upload, the light starts flashing. Anyone could give me some light here? Really need this to work.

[27 more replies]