espressif/esptool #433: Flashing with my own STM32 as serial adapter

matthewlai, 2019-05-22:
Hello!

I have a bit of an unusual problem here... I am trying to flash an ESP8266 on a ESP-11 module, using an STM32F7 on the same board as a USB to serial adapter. I can't seem to flash anything (not even stub), but if I use "--no-stub", chip_id and flash_id work. I am also able to talk to the existing firmware (an ancient AI-Thinker version from many years ago).

I need to use this strange setup to flash my chip because ESP-11 is an SMD module, and I can't easily flash it out of the PCB.

I wrote the USB-UART converter firmware for the STM32 that enumerates as a USB serial port and passes messages between USB and UART, so I have a lot of control there. I am guessing there's something wrong with how I implemented it, but I'm not sure what. I do have baud rate change implemented (and verified working with an oscilloscope), but not any of the flow control signals - are they necessary? If so, what should I do with them (what ESP8266 pins should I toggle)? I have control of CH_EN, RST, Rx, Tx, GPIO0, and GPIO1. At the moment I have a short delay after power up before grounding GPIO0, then driving CH_EN and RST to Vcc.

* Operating system: Ubuntu 18.04
* Python version: 2.7.16
* ESP hardware in use: ESP8266 on ESP-11

# Full esptool.py command line as run:
esptool.py -p /dev/ttyACM0 chip_id

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.6
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:9e:6e:80
Uploading stub...

A fatal error occurred: Timed out waiting for packet header

# What is the expected behaviour?
Stub upload succeeds.

# Do you have any other information from investigating this?
I am able to get flash_id and chip_id to work with --no-stub. I am only using "chip_id" as an example above, as write_flash doesn't work with --no-stub.

Log with --trace: https://gist.github.com/matthewlai/752789ce9b385159332b696f425cccc5

Thanks!

matthewlai, 2019-05-22:
"--before no_reset" also didn't help.

projectgus, 2019-05-23:
Hi @matthewlai ,

Thanks for the detailed description of your setup.

If the ESP8266 is resetting into download mode correctly with GPIO0, then you shouldn't need to do anything with UART flow control pins. It seems like this is the case as you get responses to short commands OK.

In the trace I see the host sends the first packet of 6160 bytes of loader code for loading into ES8266's RAM, and then times out after 3 seconds with no response from the ESP8266 ROM loader.

I also noticed in the trace that the STM32 is only sending 1-2 bytes per USB packet back to the PC. This is not a problem by itself on the host side, and in fact it may just be indicative of a very fast USB device, but something to keep in mind all the same.

The first thing you could try is increasing the 3 second `DEFAULT_TIMEOUT` value in the code to something larger, in case the STM32 is taking longer to send all the data and then relay back the response.

If this doesn't help then I recommend attaching a logic analyzer to the STM32 UART TX/RX pins, and check that all bytes which are sent from the USB host are sent by the STM32, and to see whether any bytes are coming back from the ESP8266 and somehow not being relayed to the PC.

If bytes are being dropped, it may be because some internal buffer in the STM32 is overflowing.

If all the bytes are the same and none are being dropped, maybe look for any difference in timing of data sent by the STM32 (this would be a bug in the ESP8266 ROM code, I don't know of such a bug but it's not impossible.)

matthewlai, 2019-05-23:
Hi @projectgus,

Thank you so much for your help! Your reply cleared up quite a few things.

It did turn out to be a buffer overflow issue on the STM32 - my USB receive buffer overflow wasn't handled correctly. Everything works now (as far as I can tell).

In case anyone stumbles upon this from Google later - CDC ACM flow control is done using NAK, and the OUT endpoint should be set to NAK if the internal buffer cannot accept another full size packet - 64 bytes in FullSpeed.

Thanks!

closed 2019-05-23 by matthewlai

projectgus, 2019-05-23:
Glad you got it working, @matthewlai :)

projectgus, 2020-01-30:
For anyone who stumbles across this issue. we now have an open source library which implements flashing of an ESP32 from another MCU. ESP8266 is very similar.

https://github.com/espressif/esp-serial-flasher