espressif/esptool #434: Flashing V1.7 firmware with detected flash

matthewlai, 2019-05-23:
Not sure if you are already aware of this, but it seems like there is an issue specific to V1.7 firmware - flashing works with auto-detected flash size, but the resulting firmware doesn't run (no response on UART). Adding "--flash_size 2MB-c1" works around it.

The workaround works for me, so I don't need help, and this is just a heads up.

Thanks

More details: https://github.com/espressif/ESP8266_NONOS_SDK/issues/179

* ESP hardware in use:
ESP-11 with swapped Winbond W25Q16

# Full esptool.py command line as run:
esptool.py -p /dev/ttyACM0 write_flash --flash_size "2MB-c1" 0x0 boot_v1.7.bin 0x01000 at/1024+1024/user1.2048.new.5.bin 0x1fc000 esp_init_data_default_v08.bin 0xfe000 blank.bin  0x1fe000 blank.bin

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.6
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:9e:6e:80
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Flash params set to 0x0030
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.3 seconds (effective 119.8 kbit/s)...
Hash of data verified.
Compressed 455508 bytes to 324659...
Wrote 455508 bytes (324659 compressed) at 0x00001000 in 28.9 seconds (effective 126.3 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x001fc000 in 0.0 seconds (effective 46.7 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x000fe000 in 0.0 seconds (effective 2977.0 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x001fe000 in 0.0 seconds (effective 2987.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

# What is the expected behaviour?
New firmware works.

# Do you have any other information from investigating this?
This works:
esptool.py -p /dev/ttyACM0 write_flash --flash_size "2MB-c1" 0x0 boot_v1.7.bin 0x01000 at/1024+1024/user1.2048.new.5.bin 0x1fc000 esp_init_data_default_v08.bin 0xfe000 blank.bin  0x1fe000 blank.bin

projectgus, 2019-05-27:
Hi @matthewlai ,

This is a problem with the default "autodetect" flash_size argument value if you have a 2MB flash chip and the firmware is built for "2MB-c1". The auto-detection sets it back to plain "2MB", which is different and not compatible.

Will fix ASAP.

For esptool.py v3.0 I plan to also change the default behaviour so that esptool.py doesn't change the binary image header size parameter at all, unless `--flash_size detect`  is explicitly given.

radimkarnis, 2024-06-17:
I have decided to close this issue due to the following reasons:

1. Recent versions of esptool.py (starting from v3.0) do not modify the image header by default, so `2MB-c1` will not get overwritten as long as `--fs detect` is not used.
2. There is no way to "detect" if users want to use the `2MB` or `2MB-c1` layout. It has to be manually specified.
3. There is a known and effective solution. Users can manually specify the flash size as `2MB-c1`.
4. This request hasn't popped up ever since. It is safe to say the users and SDKs got used to the correct workflow of specifying the `2MB-c1` manually when needed.

closed 2024-06-17 by radimkarnis