espressif/esptool #435: WeMos D1 Mini working at 460800 but not 921600

projectgus, 2019-05-28:
* Operating system: Linux
* Python version: 2.7
* ESP hardware in use: WeMos Mini D1 (ESP8266 + ch341 serial chip)

# Full esptool.py command line as run:

```
esptool.py --chip esp8266 --port /dev/ttyUSB0 --baud 921600 write_flash 0x0 images/fifty_kb.bin
```
(as run by test_esptool.py)

# Full output from esptool.py (please copy and paste all lines of output)

```
esptool.py v2.7-dev
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: dc:4f:22:1d:87:df
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Invalid head of packet (0xE0)
```

# What is the expected behaviour?

As noted in the [Arduino ESP8266 repo](https://github.com/esp8266/Arduino/pull/5635#issuecomment-495034459), flashing this board with 921600bps works in esptool-ck:

```
$ ./esptool -cd nodemcu -cb 921600 -v -cp /dev/ttyUSB0 -cf ~/esp/esptool-py/test/images/fifty_kb.bin
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
opening port /dev/ttyUSB0 at 921600
opening bootloader
resetting board
trying to connect
trying to connect
Uploading 51200 bytes from /home/gus/esp/esptool-py/test/images/fifty_kb.bin to flash at 0x00000000
..................................................                               [ 100% ]
starting app without reboot
closing bootloader
```

And using 460800bps serial works fine in esptool.py as well.

# Do you have any other information from investigating this?

* esptool-ck uses the auto-baud feature of the ESP8266 to establish a link at 921600 and then keeps using that rate.
* esptool.py establishes the link at 115200bps and then [switches serial to a higher rate based on the multiplier that is being used for 115200bps](https://github.com/espressif/esptool/blob/c583756c118039cfcfe256f7a3285618914d16a5/flasher_stub/stub_flasher.c#L53). This is more reliable for at least some hardware where auto-bauding at high rates sometimes results in baud mismatch. It also aims to account for any clock skew between the serial device and the ESP device's crystal.

Probably the "switch to new baud rate" has a bug, or actually works less well on some devices (or with some drivers?).

Further investigation needed. This is not high priority, as @earlephilhower notes at the linked PR esptool.py is still faster here due to the compressed upload feature (beyond a certain point, the limiting factor becomes flash erase & write times rather than serial transfer times.)

projectgus, 2019-05-28:
Interesting: `-b 1000000` works reliably on this board when `-b 921600` does not. So it may be just to do with the available clock dividers on CH341, or the way that the Linux driver is choosing the clock divider.

(cc @earlephilhower as this may be useful info.)

pfeerick, 2019-05-29:
I can also reproduce the `-b 1000000` works,` -b 921600` doesn't work behaviour on my system... for Linux anyway. For Windows, either speed worked fine. Wemos D1 Mini with CH340G UART.

Linux: ./esptool -cd nodemcu -cb 921600 -v -cp /dev/ttyUSB0 -cf ../esptool/test/images/fifty_kb.bin 
```
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
opening port /dev/ttyUSB0 at 921600
opening bootloader
resetting board
trying to connect
trying to connect
Uploading 51200 bytes from ../esptool/test/images/fifty_kb.bin to flash at 0x00000000
..................................................                               [ 100% ]
starting app without reboot
closing bootloader
```
Linux: esptool.py --chip esp8266 --port /dev/ttyUSB0 --baud 921600 write_flash 0x0 images/fifty_kb.bin
```
esptool.py v2.7-dev
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:2f:df:d6
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Invalid head of packet (0xE0)
```

Windows: python esptool.py --chip esp8266 --port COM7 --baud 921600 write_flash 0x0 test/images/fifty_kb.bin
```
esptool.py v2.7-dev
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:2f:df:d6
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 51200 bytes to 51226...
Wrote 51200 bytes (51226 compressed) at 0x00000000 in 1.0 seconds (effective 408.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Windows: python esptool.py --chip esp8266 --port COM7 --baud 1000000 write_flash 0x0 test/images/fifty_kb.bin
```
esptool.py v2.7-dev
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:2f:df:d6
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1000000
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 51200 bytes to 51226...
Wrote 51200 bytes (51226 compressed) at 0x00000000 in 0.9 seconds (effective 458.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

projectgus, 2019-05-30:
Thanks for confirming @pfeerick . Interesting both work with the Windows driver.

Oddly, the Linux driver code suggests 921600 should be perfectly accurate and 1000000 will have a small baud rate error:
https://github.com/torvalds/linux/blob/master/drivers/usb/serial/ch341.c#L157

For 921600, `factor = 1532620800 / 921600` which is 1663 (no error).

For 1000000, `factor = 1532620800 / 1000000` which is 1532.6208 so the actual baud rate will be `1532620800 / 1532` which is 1000405.2bps

It certainly merits further investigation. :)

yumkam, 2020-07-27:
FTR, recent linux kernels (5.5+) has  [some](https://github.com/torvalds/linux/commit/35714565089e5e8b091c1155517b67e29118f09d) [fixes](https://github.com/torvalds/linux/commit/7c3d02285ad558691f27fde760bcd841baa27eab) in ch341 driver w.r.t baudrate setting; I've played with backported ch341 module, it improved things a bit for me (top speed was 500k before, 576k after; still no luck with 921600/1M/1152k/1.5M; fwiw, `read_flash` was more prone to failures than `write_flash`).
(Different 8266 board - with cp2102 - seems works with 1.5M, 921600, 576k, 500k, 460800, 230400 [but NAK with 1M or 1152k])

anm, 2020-08-16:
I have exactly the same problem with a CH340G and an ESP32, including the error 0xE0, although often it instead says timeout waiting for packet content or packet header.

sblantipodi, 2021-01-01:
do they fixed the 921600 baud rate problem?
there are too many errors at 1000000

pfeerick, 2021-01-01:
I don't have any issues with 460800, which is what the Arduino ESP8266 core remaps anything above 460800. So I generally don't go above that, unless I'm on a hardware/OS combination that will take the higher baud rate... which is becoming increasingly less often as I do more and more OTA update/programming instead.

sblantipodi, 2021-01-01:
@pfeerick thanks for the answer, it's always a pleasure to read your posts on GitHub 😃 
I think that it should be something elsein Arduino ESP8266 since I'm using it but a 1M baud rate is 2 times faster than a 500K one.

I think that Arduino ESP8266 can accept values greater than 460800 now.

sblantipodi, 2021-01-01:
but is this a problem of the ESPTool?
can you drive some LEDs for example at 921600 ?

radimkarnis, 2021-04-29:
I tested this extensively with both the WeMos D1 Mini and Mini Pro (CH340G/C) and couldn't reproduce the issue on Windows and Linux. All mentioned baudrates work fine. This leads me to believe this is not an esptool issue (more likely a CH341/Linux one).
Closing this issue, feel free to reopen if you feel like further investigation is needed.

closed 2021-04-29 by radimkarnis

sblantipodi, 2021-04-30:
I have tested lots of ESP8266 boards, D1 Mini uses the CH340G that works stable with pretty every boud rate up to 1.000.000 

Most NodeMCUs uses the CP2102 and IMHO its pretty picky.  
it doesn't like 1.000.000 baud rate but it likes 921.600, it doesn't like 500.000 but it likes 460800 .

Just use D1 Mini with CH340 if you can.