espressif/esptool #440: Flash a esp-12 from a ESP32

Nathan-ma, 2019-07-02:
Hello, I'm wanna make a esp32 code to upload another board's code (esp-12). 
The bin code would be inside ESP32 and the GPIO's and UART pins necessary to made the Uart flash to happen.

My only question is, whats the protocol I need to send via uart to the esp-12 in order to accomplish my project.

The pre compiled bin file would be inside the SPI of the ESP32.

So my only request is an explanation or part of the code for me to understand how i'm gonna send the .bin to the right memory address of the esp-12 in order to flash it.

I will control flash and boot modes from the GPIO's of the ESP32 no problem, just need to know how to send the bin.

Nathan-ma, 2019-07-02:
I went back 5 years for the first commits of this git. A simplier code i'm gonna try to work with.

> Protocol
> The bootloader protocol uses SLIP framing. Each packet is begin and end with 0xC0, all occurrences of 0xC0 and 0xDB inside the packet are replaced with 0xDB 0xDC and `0xDB 0xDD?, respectively.
> 
> Inside the framing, the packet consists of a header and a variable-length body.

That's what I've found on the commits, I'm going to experiment a few codes to see if I can successefuly develop this uploader on board to another board. Any help and tips are welcome!

projectgus, 2019-07-03:
Hi @Nathan-ma ,

This wiki page should help as well:
https://github.com/espressif/esptool/wiki/Serial-Protocol

I'm going to close the issue as this isn't something we will implement as part of esptool itself, but we are considering creating an ESP-IDF example that demonstrates using the serial protocol to flash another ESP32, with the idea that this could also be ported to other host micros.

Even though you plan to flash an ESP8266, there are some threads on the ESP32 forums you may find helpful, as well:
* https://esp32.com/viewtopic.php?t=7400
* https://esp32.com/viewtopic.php?f=2&t=11296&p=45832#p45832

Angus

closed 2019-07-03 by projectgus

Nathan-ma, 2019-07-03:
Thank you Angus, The esp-idf example would be something amazing, Please keep me informed as soon as it is released!
I really need this code for some projects i'm developing

Nathan-ma, 2019-07-03:
@projectgus You know where I can find the bin  command to put Wemos boards on bootloader mode?
thank you Angus

igrr, 2020-01-29:
@Nathan-ma Sorry that it took a while, but we now have an example of flashing an ESP32 from an ESP32. Flashing of an ESP8266 will be added soon. https://github.com/espressif/esp-serial-flasher