espressif/esptool #441: esptool.FatalError: Timed out waiting for packet header

Taufiqur2706, 2019-07-03:
**I try to connect my lolin ESP8266 to blynk using this script**

/*************************************************************
  Download latest Blynk library here:
    https://github.com/blynkkk/blynk-library/releases/latest

  Blynk is a platform with iOS and Android apps to control
  Arduino, Raspberry Pi and the likes over the Internet.
  You can easily build graphic interfaces for all your
  projects by simply dragging and dropping widgets.

    Downloads, docs, tutorials: http://www.blynk.cc
    Sketch generator:           http://examples.blynk.cc
    Blynk community:            http://community.blynk.cc
    Follow us:                  http://www.fb.com/blynkapp
                                http://twitter.com/blynk_app

  Blynk library is licensed under MIT license
  This example code is in public domain.

 *************************************************************

  You’ll need:
   - Blynk App (download from AppStore or Google Play)
   - NodeMCU board
   - Decide how to connect to Blynk
     (USB, Ethernet, Wi-Fi, Bluetooth, ...)

  There is a bunch of great example sketches included to show you how to get
  started. Think of them as LEGO bricks  and combine them as you wish.
  For example, take the Ethernet Shield sketch and combine it with the
  Servo example, or choose a USB sketch and add a code from SendData
  example.
 *************************************************************/

/* Comment this out to disable prints and save space */
#define BLYNK_PRINT Serial

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "f348bdb682a047b5983f59ec53b6485f";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "TaufiqHotSpot";
char pass[] = "tidakadapassword";

void setup()
{
  // Debug console
  Serial.begin(9600);

  Blynk.begin(auth, ssid, pass);
  // You can also specify server:
  //Blynk.begin(auth, ssid, pass, "blynk-cloud.com", 80);
  //Blynk.begin(auth, ssid, pass, IPAddress(192,168,1,100), 8080);
}

void loop()
{
  Blynk.run();
  // You can inject your own code or combine it with other sketches.
  // Check other examples on how to communicate with Blynk. Remember
  // to avoid delay() function!
}
**But during Upload there is  a problem with this error code**

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), ck, 26 MHz, 40MHz, DOUT (compatible), 512K (no SPIFFS), 2, nonos-sdk 2.2.1 (legacy), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Sketch uses 279952 bytes (56%) of program storage space. Maximum is 499696 bytes.
Global variables use 28864 bytes (35%) of dynamic memory, leaving 53056 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:05:85:e1
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Traceback (most recent call last):
  File "C:\Users\har.outage.spv\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/har.outage.spv/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2694, in main
    detect_flash_size(esp, args)
  File "C:/Users/har.outage.spv/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2073, in detect_flash_size
    flash_id = esp.flash_id()
  File "C:/Users/har.outage.spv/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 567, in flash_id
    return self.run_spiflash_command(SPIFLASH_RDID, b"", 24)
  File "C:/Users/har.outage.spv/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 812, in run_spiflash_command
    old_spi_usr = self.read_reg(SPI_USR_REG)
  File "C:/Users/har.outage.spv/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 475, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
  File "C:/Users/har.outage.spv/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 332, in command
    p = self.read()
  File "C:/Users/har.outage.spv/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 277, in read
    return next(self._slip_reader)
  File "C:/Users/har.outage.spv/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 1877, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header

Taufiqur2706, 2019-07-03:
Feel free to give us the problem solver...
thanks

yuriystovbur, 2019-07-10:
I have a similar problem.

- Windows 7
- Arduino 1.8.9
- [esp8266 2.5.2 by ESP8266 Community](https://github.com/esp8266/Arduino/tree/2.5.2)
- [WIFI NodeM ESP8266, CP2102](https://robotdyn.com/wifi-nodem-esp8266-cp2102.html)

[image]

[image]

[image]

```
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: f3:e3:84:7f:62:a8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Traceback (most recent call last):
  File "...\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File ".../esp8266/2.5.2/tools/esptool\esptool.py", line 2694, in main
    detect_flash_size(esp, args)
  File ".../esp8266/2.5.2/tools/esptool\esptool.py", line 2073, in detect_flash_size
    flash_id = esp.flash_id()
  File ".../esp8266/2.5.2/tools/esptool\esptool.py", line 567, in flash_id
    return self.run_spiflash_command(SPIFLASH_RDID, b"", 24)
  File ".../esp8266/2.5.2/tools/esptool\esptool.py", line 812, in run_spiflash_command
    old_spi_usr = self.read_reg(SPI_USR_REG)
  File ".../esp8266/2.5.2/tools/esptool\esptool.py", line 475, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
  File ".../esp8266/2.5.2/tools/esptool\esptool.py", line 332, in command
    p = self.read()
  File ".../esp8266/2.5.2/tools/esptool\esptool.py", line 277, in read
    return next(self._slip_reader)
  File ".../esp8266/2.5.2/tools/esptool\esptool.py", line 1877, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header
```

Taufiqur2706, 2019-07-10:
thank you for the respons.I would try itSent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/espressif/esptool/issues/441?email_source=notifications\u0026email_token=AMQME5S5QSR5S6CIJ6CMIJ3P6V4GVA5CNFSM4H5BRNS2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZSM2EQ#issuecomment-509922578",
"url": "https://github.com/espressif/esptool/issues/441?email_source=notifications\u0026email_token=AMQME5S5QSR5S6CIJ6CMIJ3P6V4GVA5CNFSM4H5BRNS2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZSM2EQ#issuecomment-509922578",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

projectgus, 2019-07-11:
Hi @Taufiqur2706 @yuriystovbur,

Thanks for all these details. It's unusual that esptool is always failing at the same point, trying to read the flash chip id.

Can you please post a high resolution photo of your board? There might be something unusual about the flash chip on them, so I'm looking to read the flash chip model number and also to identify the board so I can try to find an identical one myself.

(I saw I can order your one from robodyn, yuri, but it may be easier for me to source it another way.)

Angus

yuriystovbur, 2019-07-12:
Hi @projectgus.

Board photos here: https://photos.app.goo.gl/Kii7zTmx5nN7cW9M7

projectgus, 2019-07-15:
@yuriystovbur Thanks. The flash chip looks like one which is commonly used (assuming genuine) but I've never seen a board layout quite like this, so I've ordered a couple to test.

yuriystovbur, 2019-07-15:
@projectgus, Thank you! We will expect results)
p.s. Maybe the chip is damaged in my case? I purchased this board on Ali - https://ru.aliexpress.com/item/32779738528.html

projectgus, 2019-07-15:
> p.s. Maybe the chip is damaged in my case? 

@yuriystovbur This is possible.

projectgus, 2019-07-23:
@yuriystovbur My two boards arrived and they both work. So I think this is probably faulty hardware, unless there is something attached to or shorting across the flash GPIOs on the board (marked as CLK, SD0, CMD, SD1, SD2, SD3). Suggest asking for a replacement for the vendor.

projectgus, 2019-10-18:
Closing due a lack of further updates, assuming this is a hardware problem.

closed 2019-10-18 by projectgus

mirospacek, 2019-10-29:
While uploading from the arduino IDE:
Hold down the FLASH button. After uploading done, release the flash button. 
Unplug USB from computer. 
Plug USB to computer back in.
The uploaded software should run.

marcosvaladez966, 2019-11-04:
Buen dia a todos, estoy trabajando con NodeMCU 8266 V3. y despues de cargar 5 o 7 programas con IDE ARDUINO, me sale este error y no puedo ejecutar nuevos programas
//
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header
//

He leeido sobre algunas soluciones como: Dejar precionado FLASH miestras se craga el programa y despues dejar de pulsar, tambien cargar un nuevo Firmware para nodemcu y empezar desde cero, "Dejo link de imagenes".
 Pero el resultado es el mismo, al parecer el modulo esta en modo ejecucion y no deja que oytro programa se carge. Alguien tiene alguan informacion o solucion al respecto, de antemano gracias!.

https://drive.google.com/drive/folders/1Q9fAlxcz7a6SuBE7R5KsR3S4Ukg_7aJM?usp=sharing

mirospacek, 2019-11-04:
Mantenga presionado el botón FLASH mientras carga. Luego desconecte la energía. Luego vuelva a aplicar el poder.

sudarkce, 2019-11-18:
esptool.FatalError: Timed out waiting for packet header :  I too was getting this error while trying to upload sketches on Espresso Lite v2.0 through UC00A usb to uart converter.  UC00A has a voltage selector switch, which is set to 5 V by default.  When I thrown this to 3V3, the sketch loaded successfully.

manibharathytu, 2020-01-05:
I had the similar issue. Scroll to last for the solution that worked for me.

**Problems I faced :**
My NodeMCU ESP8266 was running good for sometime and I was controlling things with Wi-Fi. But after some days it started to disconnect from Wi-Fi and I had to reset my ESP to reconnect to Wi-Fi and all of sudden my ESP is disconnected to my router. I tried to access the softAP running in my ESP. That was also not discoverable. I reset the  board by pressing reset button on the board and it started working. This started to  happen frequently. And one day, this issue happened I reset the board
but still not connected to my router because it forgot the saved Wi-Fi credentials. I had to reconfigure the credentials through softAP. Then this issue started to appear now and then. Everytime I had to reset and reconfigure credentials

And one day this issue happened agian, but even after resetting, I was not able to access softAP to reconfigure credentials. So I opened serial monitor. Usually it prints my logs good. It use to print some garbage line while setting up the Wi-Fi connection (I don't know why) and then it will print my logs. You can see blow the garbage line it prints when starting the board.

`rl $ܞ⸮⸮c⸮N⸮$⸮"|⸮⸮⸮⸮2⸮s⸮b⸮p⸮⸮oN⸮ lNn⸮⸮⸮cp⸮#$`{lp⸮N⸮⸮l⸮⸮cN⸮|l⸮⸮p⸮|⸮n⸮d⸮⸮ d ⸮nol`or⸮⸮⸮o"⸮${⸮⸮⸮c⸮l⸮"r#⸮⸮p{$⸮⸮n⸮
HTTP server started
192.168.1.2
0
Configuring access point...analog
analog
digi
.
.
`

But now it was continously printing that garbage. Seems like the ESP Wifi configuration library got stuck in a loop. I thought some code might got corrupt, so I tried re-uploading the code. That's when I faced this issue. When I tried to upload the console says
```
Connecting........_____....._____....._____....._____....._____....._____.....____
.
.
.

esptool.FatalError: Timed out waiting for packet header
```

Things I tried that did not work :

- Reset button
- Unplug and replug USB
- Restart Arduino IDE
- Holding the FLASH button and upload

**Solution :**
I removed the GPIO power pins connected to the circuit, now code is getting uploaded. And my board works as expected.

Why the fix worked?
I'm not sure what was the problem and how removing the connection to circuit helped. If you must know, the pins I removed was the Vin and GND for power and D4 which was connected to optocoupler and was giving PWM output to opto.

These kinds of issues makes ESP board unreliable. I'm also wondering these issues happen because of my mishandling like cutting power when its working or uploading etc.

Damiortiz, 2020-01-07:
I tried all the things I saw on the Internet.  The only solution that worked was this. 
1. Press the flash and RST button at the same time for a few seconds.
2. Release only the RST button
3. Without releasing the FLASH button upload the code with NodeMCU version 1.0

I hope someone will use it!

davefuentes, 2020-01-21:
In my case, cable, drivers, and board was ok. The issue was that I had connected vin and ground. Once I removed those pins everything worked fine

alanfelipe19, 2020-02-29:
> In my case, cable, drivers, and board was ok. The issue was that I had connected vin and ground. Once I removed those pins everything worked fine

**It was just that.** Thanks

sganzerla, 2020-03-25:
Hi, in my case, error resolved removed buzzer DR-08H.

HaegarDerWikinger, 2020-04-03:
> @yuriystovbur My two boards arrived and they both work. So I think this is probably faulty hardware, unless there is something attached to or shorting across the flash GPIOs on the board (marked as CLK, SD0, CMD, SD1, SD2, SD3). Suggest asking for a replacement for the vendor.

I had the same. connected a new HDT11 to the ESP8266 pin D8. After I disconnect it, I could download the sketch

NieVystrel, 2020-06-11:
Here some points if you have troubles with you ESP01 (idk if its considered ESP8266 but i think so):

- Did you install the drivers at the very first instant? Maybe didnt notice windows splash you a "no installed device screen". Silabs is the way (for the ORIGINAL boards). O.o

- Check the USB cable you are using, i want to sure this point connecting all the time directly in the gabinet (idk if its risky i took the challenge).

-If i understood correctly, there are chinease boards around, so the Silabs driver doesnt response nice with this fakes (in my case it looks like the driver installed correctly but Windows doesnt asociate to the board, so the doubts begun), try this Driver from sparkfun tutorial https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all this saves me because no one in all the 5 hours googling mention it.

-If you bought a USB programmer like this 
[image]
ensure to add a wire between pins mentioned in this tutorial, credits for him https://www.youtube.com/watch?v=m4fUswZSHcY. THIS IS FOR SETTING THE BOARD IN PROGRAMMING MODE so Arduino IDE can write in it.

I having very bad times configuring EVERYTHING, fighting with the INDOMITABLE Board Manager, learning about GIT, and doing stuff that never cross in front of my eyes...people who starting in this things get frustated and left this wonderful world because lack of time to research (some lazies its ok...) or the lenguage used in some topics is hard to comprehend, as suggestion to all: be kind as you want to be treated in life.

So basically this is my humble aport to someone who is newbie like me in the IoT community, i think its my impression some users doesnt have the will to help others from heart, its like always you have tear the words from they. Bah.

Two nice points of the quarentine: time to research you want and self repairing of the nature :D greetings from Mexico (sorry if i bothered you with no common syntaxis here hehe).

laxmru, 2020-07-19:
I could resolve it just choosing right port
In my case, I was connecting to that port which did not have CH340 driver.

FrankVolt, 2020-08-17:
After dropping the upload speed to 115200 the FatalError: Timed out .... was solved. I use the Arduino IDE

ibrahimqazi, 2020-08-19:
I had the same problem and found the solution:
While burning the code to board, remove any connected device, and must press and hold FLASH button until the IDE starts burning the code, after that you can release FLASH button during the burning process. Otherwise you may face problems while burning the code.

FreddieDev, 2020-11-16:
I had the same issue, this worked for me:

1. Select "NodeMCU 1.0 (ESP-12E Module)"
2. Unplug Lolin NodeMCU v3 from PC
3. Connect to PC while holding FLASH with no wires connected
4. Upload

May a couple of attempts

eneiasmg, 2021-03-18:
@FreddieDev thank you! That worked at first time!

aleydapacheco, 2022-01-03:
Hola a todos, comentarles que llegue hasta esta pagina justamente por el caso de:
 "... line 2636, in slip_reader raise 
FatalError("Timed out waiting for packet %s" % waiting_for) esptool.
FatalError: Timed out waiting for packet header esptool.
FatalError: Timed out waiting for packet header... "
Lo que soluciono mi problema fue levantar el modulo ESP8266 o NodeMCU de la placa protoboard donde estaba montado. Una vez retirado el modulo tan solo conectado son el cable USB, trate de subirlo sin la necesidad de hacer presión en ninguno de los botones y funciono. Deduzco que en algún lugar había un pin o tierra mal conectada. Saludos!!

iampingoo, 2022-01-14:
> I had the same issue, this worked for me:
> 
> 1. Select "NodeMCU 1.0 (ESP-12E Module)"
> 2. Unplug Lolin NodeMCU v3 from PC
> 3. Connect to PC while holding FLASH with no wires connected
> 4. Upload
> 
> May take a couple of attempts
Thank you for the good advice. It makes me happy. 👍 
My succeed environments about ESP8266 programming.

0. [Hardware ESP-01S](https://github.com/iampingoo/ESP-01S/blob/main/ESP-01S.jpg) 
1. [circuit](https://github.com/iampingoo/ESP-01S/blob/main/circuit(Programming).png)
2. [project settings](https://github.com/iampingoo/ESP-01S/blob/main/project_setting.png)

iampingoo, 2022-01-14:
> 

Ah-, I had been already able to set "Generic ESP8266" (No need to set NodeMCU 1.0...). 
This communication failure is occurring from Cheap Power, see below.

https://arduino-esp8266.readthedocs.io/en/latest/boards.html#:~:text=Provide%20sufficient%20power,capacitors%20is%20preferred.