espressif/esptool #442: could not erase ESP8266 flash

bifeiwo, 2019-07-05:
(Please delete any lines which don't apply.)

* Operating system: Windows 10 ( x64)
* Python version: (`python -V` to check this) 2.7.16 (64-bit)
* ESP hardware in use: ESP8266

# Full esptool.py command line as run: esptool.py --port COM3 erase_flash

# Full output from esptool.py (please copy and paste all lines of output)
D:\Python\Scripts>esptool.py --port COM3 erase_flash
esptool.py v2.6
Serial port COM3
Traceback (most recent call last):
  File "D:\Python\Scripts\esptool.py-script.py", line 9, in <module>
    load_entry_point('esptool==2.6', 'console_scripts', 'esptool.py')()
  File "D:\Python\lib\site-packages\esptool.py", line 2952, in _main
    main()
  File "D:\Python\lib\site-packages\esptool.py", line 2646, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "D:\Python\lib\site-packages\esptool.py", line 258, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "D:\Python\lib\site-packages\esptool.py", line 222, in __init__
    self._port = serial.serial_for_url(port)
  File "D:\Python\lib\site-packages\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "D:\Python\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': WindowsError(5, '\xbe\xdc\xbe\xf8\xb7\xc3\xce\xca\xa1\xa3
')

# What is the expected behaviour?

D:\Python\Scripts>esptool.py --port COM3 erase_flash
esptool.py v2.6
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: a4:cf:12:bf:54:4d
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 9.4s
Hard resetting via RTS pin...

# Do you have any other information from investigating this?
Actually, this mistake occur many times, I have also tried some solutions.
And I have found that any of  the follows code can get right expetation:
**esptool.py --port /COM3 erase_flash**
**esptool.py --port COM3: erase_flash**

What's more, if I type in esptool.py --port COM3 erase_flash in PowerShell,
I can get the right responsion

I also tried typing in command **esptool.py --port COM3 erase_flash** while I use usb to connect esp8266  with the computer and it worked correctly.

I don't know what should I do and why the mistake occurs.

projectgus, 2019-07-08:
@bifeiwo That's a strange error!

To make sure I understand: you run this command in Windows Command Prompt, you get the error above. If you run the same command in PowerShell, no error?

I don't understand this:

The root cause is an error from Windows when opening COM3: ` WindowsError(5, '\xbe\xdc\xbe\xf8\xb7\xc3\xce\xca\xa1\xa3`. Windows Error 5 is [Access Denied](https://docs.microsoft.com/en-us/windows/win32/debug/system-error-codes--0-499-).

Do you some antivirus or security software installed that might be interfering with attempts to access th serial ports?

bifeiwo, 2019-07-08:
Yes, as what I said above, if I use powershell or I add some symbols to the command, I can run it correctly.
And after my test, I found that the CMD was wrong with the first nine COM port  ( COM1-COM9), but there was no problem at the else COM ports.
I don not know if this is my computer problem, I think CMD may have no permission on these 9 COM ports on my computer, because I can use any other software to  open the serial port and there is not any problem.

bifeiwo, 2019-07-08:
yes, as what I said above, if I use powershell or I add some symbols to the command, I can run it correctly.
And after my test, I found that the CMD was wrong with the first nine COM port ( COM1-COM9), but there was no problem at the else COM ports.
I don not know if this is my computer problem, I think CMD may have no permission on these 9 COM ports on my computer, because I can use any other software to open the serial port and there is not any problem.
Thanks for your help.

projectgus, 2019-07-09:
Hi @bifeiwo ,

Yes, I don't think this is anything esptool is doing - the operating system is telling us "access denied" so it's something in the OS.

As a test, if you run cmd.exe "as administrator" can you run esptool successfully then?

bifeiwo, 2019-07-09:
No, I seems that there is some problems about cmd. I 'll try to ask for help from windows

closed 2019-07-09 by bifeiwo

alaj, 2020-05-02:
I have same problem when I use CMD. But git bash is good.

drevil31, 2020-07-23:
You get the same error if you still have that port open with for example PuTTy.
So maybe something hidden is connected and blocks windows from access.