espressif/esptool #445: esptool 2.6 binaries on windows

denisfou, 2019-07-14:
Hello, 

i can't have a working ESP device since esp2.6 (provided by ESP on arduino IDE by community package 2.5.1 and later)

* Operating system: Windows 10 pro
* ESP hardware in use: ESP01 or nodemcu v3
[image]
[image]
[image]
[image]
[image]
[image]
[image]

# Full esptool.py command line as run: 
C:\Users\denis\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1/python C:\Users\denis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py --chip esp8266 --port COM3 --baud 115200 --trace version --end --chip esp8266 --port COM3 --baud 115200 --trace write_flash 0x0 C:\Users\denis\AppData\Local\Temp\arduino_build_52442/test.ino.bin --end 

# Full output from esptool.py (please copy and paste all lines of output)
Sketch uses 262480 bytes (52%) of program storage space. Maximum is 499696 bytes.
Global variables use 26664 bytes (32%) of dynamic memory, leaving 55256 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:84:80:25
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0320
Compressed 266640 bytes to 194386...

Writing at 0x00000000... (8 %)
Writing at 0x00004000... (16 %)
Writing at 0x00008000... (25 %)
Writing at 0x0000c000... (33 %)
Writing at 0x00010000... (41 %)
Writing at 0x00014000... (50 %)
Writing at 0x00018000... (58 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (75 %)
Writing at 0x00024000... (83 %)
Writing at 0x00028000... (91 %)
Writing at 0x0002c000... (100 %)
Wrote 266640 bytes (194386 compressed) at 0x00000000 in 17.2 seconds (effective 123.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

# What is the expected behaviour?
the device must work, but it does nothing.

# Do you have any other information from investigating this?
Version of esptool provided by package 2.5.0 of ESP8266 community package was working

# Is there any other information you can think of which will help us reproduce this problem?
install ide on windows 10, update package ESP to 2.51 ou 2.52 => this doesn't work. Binaries are uploaded but the device is not working. 

Same issues on ESP01  and nodemcu devices.

my code is : 

void setup() 
{
  Serial.begin(115200);
}

void loop() 
{
  Serial.printf( "ESP8266 Test\n" );
  delay(1000);
}

projectgus, 2019-07-15:
Hi @denisfou ,

The output from esptool suggests that it wrote the data to flash OK, and it was read back and verified correctly. I think probably this is a problem when the Arduino firmware tries to start from the chip.

If you look at the serial output of the chip booting after reset with baud rate 74880, do you see any output?

One thing I noticed in the `write_flash` command is that the command line options for flash mode, flash size, and flash speed are not being passed to esptool.py by Arduino. This means in particular that esptool is setting the flash size param to 1MB (as this is the size detected by esptool), but in Arduino IDE the size is configured as 512KB. This may cause failure to flash.

You could try changing "Flash Size" to 1MB in Arduino IDE and see if this makes any difference. If so, this is a bug in ESP8266 Arduino (not setting the parameter) not a bug in esptool.

denisfou, 2019-07-15:
Thank you, I will try. 
But i still wonder why version 2.5.0 works and 2.5.2 doesn't with the same IDE.

projectgus, 2019-07-16:
@denisfou ESP8266 Arduino project switched from esptool-ck flasher tool in 2.5.0 to using esptool.py in 2.5.2. So 2.5.0 is using a totally different tool, different command line arguments, etc.

denisfou, 2019-08-07:
> 
> 
> @denisfou ESP8266 Arduino project switched from esptool-ck flasher tool in 2.5.0 to using esptool.py in release 2.5.1. So 2.5.0 is using a totally different tool, different command line arguments, etc.

sorry, for late reply, i was in holidays.

If tool for 2.5.1 is different maybe agruments produced by windows adruino IDE are not good.

denisfou, 2019-08-08:
Just tested on Centos7 64 bits and arduino ide 1.8.9. 
I have exactly the same problem :(

**2.5.0 works fine and 2.5.1, 2.5.2 fail.**

All ESP devices based using **ESP8266 community board manager** greater than 2.5.0 have the problem.

I don't know where to open an issue for that.

projectgus, 2019-08-09:
Hi @denisfou ,

Please open an issue with the ESP8266 Arduino project, here: https://github.com/esp8266/Arduino/issues

All esptool.py will do is write the bytes to flash that it is told to write. If what is written fails to boot the device, it's up to the caller of esptool.py to correct this (in this case, the caller is Arduino).

Angus

closed 2019-08-09 by projectgus

denisfou, 2019-08-09:
ok i found the solution.

in the file platform.txt of esp8266 board distribution at the line **tools.esptool.upload.pattern** , the following argument is missing : 

**--after soft_reset**

the good line become :

tools.esptool.upload.pattern="{cmd}" "{runtime.platform.path}/tools/upload.py" --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" "{upload.verbose}" {upload.erase_cmd} --end --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" "{upload.verbose}" **--after soft_reset write_flash** 0x0 "{build.path}/{build.project_name}.bin" --end

denisfou, 2019-08-09:
Thank you  :)

I found a solution here.

https://github.com/esp8266/Arduino/issues/6296

> Hi @denisfou ,
>
> Please open an issue with the ESP8266 Arduino project, here: https://github.com/esp8266/Arduino/issues
>
> All esptool.py will do is write the bytes to flash that it is told to write. [...]