espressif/esptool #448: Finding flash Address

realgam3, 2019-07-28:
Hi,
I'm trying to parse my ESP8266v2 firmware
and I have noticed this line:
irom_segment.addr = 0  # for actual mapped addr, add ESP8266ROM.IROM_MAP_START + flashing_addr + 8
// https://github.com/espressif/esptool/blob/1b310921cf5b6eb14f3d59aeba5ae84ceda6e18a/esptool.py#L1583

how do I find the flashing_addr?

projectgus, 2019-08-01:
Hi @realgam3 ,

This comment means that the "real" mapping address in the virtual address is determined by where the .bin file is written in the flash (physical address).

So if the .bin file is written to flash at offset 0x0, it is 0x0. If it is written at offset 0x100, it is 0x100, etc.

The documentation for whatever SDK the firmware was generated from should have this info.

projectgus, 2019-10-18:
Closing due a lack of further information.

closed 2019-10-18 by projectgus