espressif/esptool #449: missing --before no_reset option

tatsutaigu, 2019-08-01:
I'm using ESP-IDF for ESP32.
esptool.py v2.7-dev ignores `Before flashing (No reset)` on menuconfig(sdkconfig).

projectgus, 2019-08-01:
Hi @tatsutaigu ,

What OS do you use and what development hardware?

Angus

tatsutaigu, 2019-08-05:
Hi
I use Windows 7 with [esp-idf-tools-setup-2.0.exe](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#esp-idf-tools-installer)

And, I use esp-idf master branch.
I tied `idf.py build`.
message is here.
> Project build complete. To flash, run this command:
> ..\..\..\..\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --after no_reset write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\ble_mesh_node.bin
> 
> or run 'idf.py -p (PORT) flash'

I set `Before flashing (No reset)` on menuconfig(sdkconfig), but I could not find `--before no_reset` option above message.
Then 'idf.py -p (PORT) flash', it seems `--before no_reset` is missing because reset pulse always was generated.
I could not determine the problem either esptool.py, esp-idf or idf.py.

projectgus, 2019-08-05:
Thanks for the extra details, @tatsutaigu .

I think this problem is in ESP-IDF as the `--before no_reset` argument is not added, as you note.

If you copy the command and then add `--before no_reset` near the beginning of the command line, does everything work?

tatsutaigu, 2019-08-05:
Hi, I could not run command as below.

```
..\..\..\..\components\esptool_py\esptool\esptool.py -p COM20 -b 460800 --before no_reset --after no_reset write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\ble_mesh_node.bin
Pyserial is not installed for C:\Python27\python.exe. Check the README for installation instructions.
Traceback (most recent call last):
  File "C:\ESP-IDF_toolchain\esp-idf\components\esptool_py\esptool\esptool.py", line 37, in <module>
    import serial
ImportError: No module named serial
```
`idf.py -p COM20 flash` can flash, but `--before no_reset` is ignored.
idf.py generates reset signal.

closed 2019-11-18 by espressif-bot