espressif/esptool #450: Can't upload program to Wemos D1 mini via USB

delta010, 2019-08-01:
* Operating system: Linux
* Python version: Python 2.7.16+
* ESP hardware in use: Wemos D1 mini

# Full esptool.py command line as run: 

esptool --port /dev/ttyUSB0   write_flash -fs 4MB -fm dout 0x0 sonoff.bin

# Full output from esptool.py (please copy and paste all lines of output)

esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/home/a/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "/home/a/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 2653, in main
    esp.connect(args.before)
  File "/home/a/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

# What is the expected behaviour?

successfully flashed

# Do you have any other information from investigating this?

while using the USB port in the Wemos D1 mini the esptool.py produce the above error But using an FTDI programmer the Flash completed without any error message 

# Is there any other information you can think of which will help us reproduce this problem?

no

projectgus, 2019-08-02:
Hi @delta010,

It's possible the automatic reset circuit in the WeMos D1 Mini is not working properly.

* If you manually short GPIO0 to GND while running esptool (can do this with a pair of pointy tweezers, or a breadboard jumper wire, for example) then does flashing complete successfully?
* If you run esptool and then quickly press and release the "Reset" button on the D1 Mini while esptool says "Connecting", does flashing complete successfully? May need to try this step more than one.

delta010, 2019-08-02:
Both methods are not working. It shows the same error

projectgus, 2019-10-18:
Hi @delta010 ,

Sorry for the long delay between replies.

Given it works from the external FTDI, my guess is that there's a problem with the USB/serial chip on the Wemos board - probably a cold solder joint or similar.

If the board is in a breadboard or anything else is connected then it may be wroth removing the connections, and also swapping the USB port and USB cable, just to be certain.

radimkarnis, 2020-12-08:
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-08 by radimkarnis