espressif/esptool #453: no Upload possible with esptool v2.7

hasenradball, 2019-08-08:
(Please delete any lines which don't apply.)

* Operating system: WIN10
* Python version: 3.7
* ESP hardware in use: ESP8266

# When usind Adruino 1.8.9 I get the following error during flash with esptool v 2.7:

```
"C:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python\\3.7.2-post1/python" "C:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/signing.py" --mode sign --privatekey "D:\\06-1_Arduino\\ESP8266\\ESP8266_Messknecht_Temp_Hum_v1_6_work/private.key" --bin "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_550954/ESP8266_Messknecht_Temp_Hum_v1_6_work.ino.bin" --out "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_550954/ESP8266_Messknecht_Temp_Hum_v1_6_work.ino.bin.signed"
Bibliothek ESP8266WiFi in Version 1.0 im Ordner: C:\Users\Frank\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi  wird verwendet
Bibliothek DHT_sensor_library_for_ESPx in Version 1.0.9 im Ordner: D:\06-1_Arduino\libraries\DHT_sensor_library_for_ESPx  wird verwendet
"C:\\Users\\Frank\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-size" -A "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_550954/ESP8266_Messknecht_Temp_Hum_v1_6_work.ino.elf"
Der Sketch verwendet 318828 Bytes (30%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.
Globale Variablen verwenden 31400 Bytes (38%) des dynamischen Speichers, 50520 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.

usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument --chip/-c: invalid choice: 'a' (choose from 'auto', 'esp8266', 'esp32')
esptool: error: argument --chip/-c: invalid choice: 'a' (choose from 'auto', 'esp8266', 'esp32')
```

projectgus, 2019-08-09:
Hi @hasenradball ,

It looks like the Arduino IDE is generating an invalid command line for esptool.py. We can't fix this from the esptool.py side.

Please report to the ESP8266 Arduino project, here:
https://github.com/esp8266/Arduino/issues

closed 2019-08-09 by projectgus

hasenradball, 2019-08-09:
Hi thanks for your answer,
Yes I think the problem is located at the arduino core.

Sorry for the wrong allocation.