espressif/esptool #454: The problem is the firmware chip esp8285 sonoff

Dionisgr, 2019-08-09:
Good day! There are prablemy firmware Sonoff RF P2 power V. 1.0. Esp8285 chip. A dump of the firmware is read fine, but when trying to flash is displayed: 
Uploading stub...
A fatal error occurred: Failed to write to target RAM (result was 0107) 
How can this be fought?
esptool.py v2.6
Python 2.7.10
Mac OS X 10.13.6

projectgus, 2019-08-14:
Hi @Dionisgr ,

Can you try lowering the baud rate with the `--baud` option?

To allow this to be debugged properly, please provide the full esptool.py command lines (for both the read and the write commands) and the full output from the failing command.

Thanks,

Angus

projectgus, 2019-10-18:
Closing due to a lack of updates.

closed 2019-10-18 by projectgus