espressif/esptool #455: Flashing ESP8266 with Arduino UNO error: Timed out waiting for packet header

MimicJojo, 2019-08-14:
Hi I have tried many things now to flash the ESP8266. Nothing works for me. I have also checked many open discussions on the Timed Out error but could't solve it with anything proposed. 
My long-term goal is to erase all memory and then run micro python on it to use in my robot. The ESP shall then read sensor data and display it live on a website. I am building the robot for my PhD and would highly appreciate any advice how I can succeed in flashing the ESP. Instead of using the Arduino IDE an c++ I'd like to build up everything using python to later apply machine learning as well.

* Operating system: Windows 7 64-bit
* Python version: Python3.7.4
* ESP hardware in use: ESP8266ex

# Full esptool.py command line as run:
`esptool.py -b 74880 -port COM6 erase_flash`
I have tried standard baud rate and 9600 as well...

# Full output from esptool.py (please copy and paste all lines of output)
`C:\Users\deathstar\AppData\Local\Programs\Python\Python37-32\Lib\site-packages>esptool.py -b 74880 -port COM6 erase_flash
esptool.py v2.5.0
Serial port COM6
Connecting........_____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header`

# What is the expected behaviour?
successful flash

# Do you have any other information from investigating this?
Here's my wiring set-up. The esptool version is 2.5.0. 
I have also tried: 

- [https://create.arduino.cc/projecthub/pratikdesai/flash-firmware-on-esp8266-esp-01-module-e1f758?f=1#comments](url)
with the result: connecting... Failed to connect

- [https://docs.micropython.org/en/latest/esp8266/tutorial/intro.html](url)
that's how I began, but then landed here on gitHub

- [https://cordobo.com/2300-flash-esp8266-01-with-arduino-uno/](url)
but here I don't get any response in Arduino IDE, probably no AT firmware pre-installed?
Wiring:
[image]

[image]

[image]

[image]

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2019-08-14:
The ground of the Arduino board needs to be connected to the ground of the ESP-01, or there is no common reference voltage.

MimicJojo, 2019-08-14:
still the same error...

faithhunja, 2019-08-14:
Once it starts to show "Connecting...", release the boot button on your ESP8266. Don't wait for too long before you release the button. If it still refuses, try again.
This worked for me with the ESP32.

MimicJojo, 2019-08-15:
resolved!
1st: used a different Power Supply than a battery ---> used another arduino (UNO) plugged into PC (3.3V pins).
2nd: I wasn't in bootmode yet, had to connect PIN 6 (EXT_ESTB) to 3.3V before powering up and flashing. 
Erase_flash now worked. I'll try next to get micropython onto that ESP.

closed 2019-08-15 by MimicJojo

projectgus, 2019-08-15:
@MimicJojo Glad you got it working!