espressif/esptool #456: Feature request: Burn multiple fuses at once

HalexRomagno, 2019-08-14:
The script restarts the handshake/connection process for every command. 
It would be nice (and faster in production programming) to bundle several fuses in one burn_efuse command.

projectgus, 2019-10-18:
Hi @HalexRomagno ,

Thanks for this suggestion. At the moment the only way that this is possible is by writing some custom Python to call into espefuse.py, and even this is not well supported.

Will see what we can do to support this.

Angus

HalexRomagno, 2019-10-18:
Hi @projectgus ,

Thank you for the response!
It would definitely be a worthwhile improvement for mass production programming.
We are currently burning 14 fuses which totals more than 30 seconds per unit.

Best,
Halex

KonstantinKondrashov, 2020-04-24:
Hi @HalexRomagno ! 
It was implemented cb5297f5bf4ad0f261747f29f48f7a8877e1683a (see the description of commit)
You can use the new commands like:

```
./espefuse.py -c esp32 burn_block_data \
BLK0 ~/esp/efuse/real/esp32_2/blk0.bin \
BLK1 ~/esp/efuse/real/esp32_2/blk1.bin \
BLK2 ~/esp/efuse/real/esp32_2/blk2.bin \
BLK3 ~/esp/efuse/real/esp32_2/blk3.bin

espefuse.py -c esp32 burn_key \
BLK1 ~/esp/efuse/real/esp32_1/blk1.bin \
BLK2 ~/esp/efuse/real/esp32_1/blk2.bin

espefuse.py -c esp32 burn_efuse \
BLK3_PART_RESERVE 1 \
DISABLE_SDIO_HOST 1 \
...
```

Added the new cmd to burn some bits: espefuse.py -c esp32 burn_bit BLK3 0 1 2 4 8 16 32 64 96 128 160 192 224 255

HalexRomagno, 2020-04-27:
Hi Konstantin,

I appreciate the work, thank you!
That will speed up the burning process quite a bit. I'll be testing the new feature in the coming weeks.

Halex

HalexRomagno, 2020-05-01:
Hi @KonstantinKondrashov ,
I'm currently implementing the batch functionality and I am running into what seems to be a formatting problem. What am I doing wrong?

Command:

> --port COM3  --before default_reset --do-not-confirm burn_efuse \ DISABLE_DL_ENCRYPT 1 \ JTAG_DISABLE 1

Response:

> usage: espefuse burn_efuse [-h]
>                            {WR_DIS,RD_DIS,FLASH_CRYPT_CNT,MAC,XPD_SDIO_FORCE,XPD_SDIO_REG,XPD_SDIO_TIEH,CLK8M_FREQ,SPI_PAD_CONFIG_CLK,SPI_PAD_CONFIG_Q,SPI_PAD_CONFIG_D,SPI_PAD_CONFIG_HD,SPI_PAD_CONFIG_CS0,FLASH_CRYPT_CONFIG,CHIP_VER_REV1,CHIP_VER_REV2,BLK3_PART_RESERVE,CHIP_VERSION,CHIP_PACKAGE,CODING_SCHEME,CONSOLE_DEBUG_DISABLE,DISABLE_SDIO_HOST,ABS_DONE_0,ABS_DONE_1,JTAG_DISABLE,DISABLE_DL_ENCRYPT,DISABLE_DL_DECRYPT,DISABLE_DL_CACHE,KEY_STATUS,ADC_VREF,BLK1,BLK2,BLK3}
>                            [new_value]
> espefuse burn_efuse: error: argument efuse_name: invalid choice: '\\' (choose from 'WR_DIS', 'RD_DIS', 'FLASH_CRYPT_CNT', 'MAC', 'XPD_SDIO_FORCE', 'XPD_SDIO_REG', 'XPD_SDIO_TIEH', 'CLK8M_FREQ', 'SPI_PAD_CONFIG_CLK', 'SPI_PAD_CONFIG_Q', 'SPI_PAD_CONFIG_D', 'SPI_PAD_CONFIG_HD', 'SPI_PAD_CONFIG_CS0', 'FLASH_CRYPT_CONFIG', 'CHIP_VER_REV1', 'CHIP_VER_REV2', 'BLK3_PART_RESERVE', 'CHIP_VERSION', 'CHIP_PACKAGE', 'CODING_SCHEME', 'CONSOLE_DEBUG_DISABLE', 'DISABLE_SDIO_HOST', 'ABS_DONE_0', 'ABS_DONE_1', 'JTAG_DISABLE', 'DISABLE_DL_ENCRYPT', 'DISABLE_DL_DECRYPT', 'DISABLE_DL_CACHE', 'KEY_STATUS', 'ADC_VR
> EF', 'BLK1', 'BLK2', 'BLK3')

HalexRomagno, 2020-05-01:
I am using Windows CMD. Here's the complete string if that can help.

> C:\Users\hromagno>cmd /c ""C:\Python34\Scripts\espefuse.py" --port COM3 burn_efuse \ DISABLE_DL_ENCRYPT 1 \ JTAG_DISABLE 1 "

HalexRomagno, 2020-05-04:
@KonstantinKondrashov User error. 
I wrongly thought this was an official release so my esptool version was 2.8 .
I'll wait for an official release before deploying this in production.

Thanks!

closed 2020-05-04 by HalexRomagno