espressif/esptool #457: ESP8266 NODEMCU V.3 LoLin esptool.FatalError: Timed out waiting for packet header

OmerBKS, 2019-08-27:
I bought a nodemcu today
I'm trying to load my first code
İt's blink but I Take error pls help me 

```
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:c4:b5:06
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Traceback (most recent call last):
  File "C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/Administrator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2694, in main
    detect_flash_size(esp, args)
  File "C:/Users/Administrator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2073, in detect_flash_size
    flash_id = esp.flash_id()
  File "C:/Users/Administrator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 567, in flash_id
    return self.run_spiflash_command(SPIFLASH_RDID, b"", 24)
  File "C:/Users/Administrator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 812, in run_spiflash_command
    old_spi_usr = self.read_reg(SPI_USR_REG)
  File "C:/Users/Administrator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 475, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
  File "C:/Users/Administrator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 332, in command
    p = self.read()
  File "C:/Users/Administrator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 277, in read
    return next(self._slip_reader)
  File "C:/Users/Administrator/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 1877, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header
```

what should I do
please help me guys 
and sorry for bad english :D

CyanoFresh, 2019-10-06:
same problem

projectgus, 2019-10-18:
Closing due to a lack of updates. Most likely this kind of error indicates a hardware problem or a serial communications issue (such as ground not connected between the USB/serial adapter and the ESP32, etc).

closed 2019-11-27 by projectgus