espressif/esptool #458: License incompatibilities (current esptool source undistributable?)

paravoid, 2019-08-29:
My understanding is:
- The entirety of esptool (Python & C code) is licensed under the GPLv2+ (i.e. with an `or later` clause)
- esptool includes flasher_stub, a small helper in C that is intended to run into the ESP chips, and thus requires (among others) the Espressif ESP8266 Non-OS SDK and the ESP-IDF SDK to build
- For convenience, pre-compiled stub codes are included in binary form in the Python source (compressed and base64-encoded)
- The Espressif ESP8266 Non-OS SDK is licensed under a proprietary license ("Espressif MIT"). This is a license based on the MIT license but placing additional restrictions (`for use on ESPRESSIF SYSTEMS ESP8266 only`).
- The ESP-IDF (as well as the ESP8266 RTOS SDK) are licensed under the Apache 2.0 license

This results in the following issues:
1) The Espressif MIT license is incompatible with (all versions of) the GPL, as the former places "further restrictions" to the terms of the GPL. Code that combines the two is undistributable, and that includes the pre-built stub code as shipped in esptool.py
2) The Apache2.0 license is incompatible with the GPLv2, but compatible with the GPLv3. This is not a problem per se: if (1) were not an issue, the stub code would make esptool effectively GPLv3. This could be perhaps further clarified in the documentation?

Note that a couple of weeks ago I filed https://github.com/espressif/ESP8266_NONOS_SDK/issues/259 asking for the Non-OS SDK to be relicensed as well. That would be the ideal resolution for issue (1).

projectgus, 2019-08-30:
That's an interesting idea that I hadn't considered before.

In the case of esptool flasher stub, our use of the Non-OS SDK and ESP-IDF is limited to including some register header files. So probably our easiest option to ensure compliance here is to re-license the register headers we need under MIT License and include them in the flasher stub directory (we could use the versions from the ESP8266 RTOS SDK, but as you point out this may create some other compatibility issues).

I realise that probably doesn't help you achieve the greater goal of putting the Non-OS SDK under an OSI approved license, you have my sympathies about that.

closed 2019-10-22 by antmak

paravoid, 2019-11-11:
So... this still requires a libc (for `stdlib.h`, `stdint.h`, `string.h` etc.) which AIUI is supposed to be a patched version of newlib as included in the SDK (or one of the many newlib-xtensa forks).

In a Debian buster system, I tried building (after a `apt install gcc-xtensa-lx106 binutils-xtensa-lx106 libnewlib-dev`) with this patch:
```diff
diff --git a/flasher_stub/Makefile b/flasher_stub/Makefile
index 0b45fd9..c443d84 100644
--- a/flasher_stub/Makefile
+++ b/flasher_stub/Makefile
@@ -62,9 +62,10 @@ all: $(STUB_PY)
 $(BUILD_DIR):
        $(Q) mkdir $@
 
-CFLAGS = -std=c99 -Wall -Werror -Os \
+CFLAGS = -std=c99 -Wall -Os \
          -mtext-section-literals -mlongcalls -nostdlib -fno-builtin -flto \
-         -Wl,-static -g -ffunction-sections -Wl,--gc-sections -Iinclude -Lld
+         -Wl,-static -g -ffunction-sections -Wl,--gc-sections -Iinclude -Lld \
+         -D__IEEE_LITTLE_ENDIAN -I/usr/include/newlib
 
 $(STUB_ELF_8266): $(SRCS) $(SRCS_8266) $(BUILD_DIR) ld/stub_8266.ld | Makefile
        @echo "  CC(8266)   $^ -> $@"
@@ -78,7 +79,7 @@ $(STUB_ELF_32S2): $(SRCS) $(BUILD_DIR) ld/stub_32s2.ld
        @echo "  CC(32S2)   $^ -> $@"
        $(Q) $(CROSS_32S2)gcc $(CFLAGS) -DESP32S2 -Tstub_32s2.ld -Wl,-Map=$(@:.elf=.map) -o $@ $(filter %.c, $^)
 
-$(STUB_PY): $(STUB_ELF_8266) $(STUB_ELF_32) $(STUB_ELF_32S2) wrap_stub.py
+$(STUB_PY): $(STUB_ELF_8266) wrap_stub.py
        @echo "  WRAP $^ -> $@"
        $(Q) $(WRAP_STUB) $(filter %.elf,$^) $@
 
```
…and it builds, but I have no idea yet if it works.

projectgus, 2019-11-11:
Hi @paravoid ,

I guess there are some differences between the toolchains we use and the lx106 toolchain packaged in Debian.

The toolchain we use for ESP8266 is: https://github.com/espressif/esptool/blob/master/.travis.yml#L16

> $ xtensa-lx106-elf-gcc --version
> xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-92-g8facf4c) 5.2.0

Specifically [the Debian package is based on 8.3.0 not 5.2.0](https://packages.debian.org/source/buster/gcc-xtensa-lx106), which was a surprise. AFAIK this is the only binary gcc8 ESP8266 toolchain being widely distributed.

We had accidentally not pushed crosstool source for our latest 5.2.0 toolchain update (you're the first person to point this out), sorry about that. It can be found here now:
https://gitlab.espressif.cn:6688/idf/crosstool-NG/tree/esp8266-1.22.x

> I have no idea yet if it works.

You can test that locally built binaries built match the binaries embedded in esptool with the `compare_stub.py` script (Travis does this as part of a CI check). I'm fairly certain gcc8 will produce different binaries though, so not that useful by itself.

The ESP32 build uses gcc 8 though, so in general the differences should be minor.

* Happy to merge PRs that fix warnings reported by gcc 8 if you have any (I noticed the `-Werror` removal in the patch).
* Would also be happy to merge PRs that added some general `EXTRA_CFLAGS`, `EXTRA_CFLAGS_8266`, `EXTRA_CFLAGS_32`, `EXTRA_CFLAGS_32S2` type variables to allow overriding these paths externally so that the stub build can support other toolchains without needing patches.
* Otherwise, at least some of these flags look like bugs in the debian toolchain package (if built for newlib then `-D__IEEE_LITTLE_ENDIAN` should be in the compiler headers already, and I don't understand why you have to push a custom include directory to include libc headers). Maybe worth reporting to the package maintainer, but also possible I just don't understand how this works from the Debian side. 
* Long term we do have a plan to release our own updated gcc for ESP8266 (actually work is underway to retool the xtensa arch support so that one toolchain binary can target multiple chips), but I don't have an ETA for this yet - not soon I'm afraid. Also unsure how that would/could coordinate with Debian toolchain policies.

paravoid, 2020-01-03:
Thanks for the response and sorry for the (very) late reply! I filed #499 for the GCC warnings/errors. As for the odd GCC version... the Debian package is tracking the upstream GCC -- and it actually has been bumped to 9.2.1 since :)

The #defines and errors above was my attempt to build with the stock newlib (which was obviously a bit dirty and probably wouldn't work).

However, in the meantime Keith Packard's [picolibc](https://github.com/keith-packard/picolibc) got support for the ESP8266, as of v1.3 which was released... 3 days ago (and pending inclusion in Debian) :) Said support was introduced by @u1f35c, who is also the maintainer in Debian for gcc-xtensa-lx106 :)

I just tried a build of gcc-xtensa-lx106 9.2.1 + picolibc 1.3 and apart from a tiny patch to disable the ESP32 builds, plus another one to remove `-Werror` (because of #499), it seems to have built fine and with no other warnings/errors. I'll give it a try on a dev board soon and see if it works and will file separate bugs if any issues arise. HTH & thanks for all the support.

paravoid, 2020-01-03:
> I just tried a build of gcc-xtensa-lx106 9.2.1 + picolibc 1.3 and apart from a tiny patch to disable the ESP32 builds, plus another one to remove `-Werror` (because of #499), it seems to have built [...]

FYI -- this is tested and works. We're not very far from shipping a working version of esptool in Debian!

u1f35c, 2020-01-03:
(I'm not sure if here is the right place to make contact, @projectgus, but I'm the Debian maintainer for the ESP8266 GCC/binutils packages. As you say we're not distributing the 5.2 or earlier versions that are common, but instead tracking the GCC version that's in Debian generally and applying the ESP8266 overlays. I'm using these for development myself with the nonOS SDK and haven't had problems, and it seems a better solution that carrying around ancient compiler versions. I'd be delighted with a solution that let me build a single package that could target ESP32 + ESP8266 and I'm happy to do whatever I can to get as many of the pieces required to work on these devices easily available in Debian.)

projectgus, 2020-01-05:
HI @u1f35c,

Great to make contact!

> instead tracking the GCC version that's in Debian generally and applying the ESP8266 overlays. I'm using these for development myself with the nonOS SDK and haven't had problems, and it seems a better solution that carrying around ancient compiler versions

That's great, I can see the reasoning. I guess the only concern to flag is that noone at Espressif has tested these toolchain versions yet, and there's not a huge amount a lot of public xtensa-gcc development outside of Espressif. So yourself and other Debian toolchain users might find some new bugs (most likely problems in SDK code not the toolchain, but possibly both). As I said above, I'm very happy to patch problems in esptool that only show up in newer gcc even if we don't support that version yet.

>  I'd be delighted with a solution that let me build a single package that could target ESP32 + ESP8266 and I'm happy to do whatever I can to get as many of the pieces required to work on these devices easily available in Debian.)

That's great, thanks for the offer of help. I think the development is still in a relatively early stage, but I'll let you know as soon as there's anything to speak of.

Have found your "noodles-blog" email address from your blog as well, will send an email there just so we have a second communications channel.