espressif/esptool #459: Flashing ESP32-PICO-D4 fails when using --no-stub option

ash-microconnect, 2019-09-05:
I'm having trouble programming an ESP32-PICO-D4 on a custom PCB if I use the '--no-stub' option. Without it, flashing works perfectly. I'm aware that it it's more than likely a hardware issue with my circuit, but before I go too far down that rabbit hole, I wanted see if someone else can perform the same test I am and whether they have the same trouble.

Using a clean install of ESP IDF v3.2.2, I compiled the 'hello_world' example and use that build as the basis for performing "normal and "no stub" flash operations. For simplicity, I'm only flashing the bootloader binary (though the same outcome occurs if I also try to flash the partitions and app). The two commands and their output are shown below. I've also tried with and without the compression ("-z") option.

Also attached is the trace output of the no-stub operation.

* Operating system: Windows 10
* Python version:  2.7.15
* ESP hardware in use: ESP32-PICO-D4 (rev 1) on custom PCB

# Full esptool.py command line as run:
With stub:
`esptool.py --chip esp32 --port COM9 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 build/bootloader/bootloader.bin`

Without stub:
`esptool.py --no-stub --chip esp32 --port COM9 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 build/bootloader/bootloader.bin`

# Full output from esptool.py
With stub:
```
ash@DESKTOP-JMV6SMR MINGW32 /c/work/esp/esp-idf/examples/get-started/hello_world
$ esptool.py --chip esp32 --port COM9 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 build/bootloader/bootloader.bin
esptool.py v2.6
Serial port COM9
Connecting.....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC: d8:a0:1d:51:84:fc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 23072 bytes to 13679...
Wrote 23072 bytes (13679 compressed) at 0x00001000 in 1.2 seconds (effective 149.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Without stub:
```
ash@DESKTOP-JMV6SMR MINGW32 /c/work/esp/esp-idf/examples/get-started/hello_world
$ esptool.py --no-stub --chip esp32 --port COM9 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 build/bootloader/bootloader.bin
esptool.py v2.6
Serial port COM9
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC: d8:a0:1d:51:84:fc
Enabling default SPI flash mode...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Erasing flash...
Flash params set to 0x0220
Compressed 23072 bytes to 13679...

A fatal error occurred: Timed out waiting for packet header
```

# Do you have any other information from investigating this?
Attached is the trace output generated using:
`esptool.py --trace --no-stub --chip esp32 --port COM9 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 build/bootloader/bootloader.bin >no-stub-trace.txt`
[no-stub-trace.txt](https://github.com/espressif/esptool/files/3577123/no-stub-trace.txt)

projectgus, 2019-09-11:
Hi @ash-microconnect ,

Thanks for the detailed report. You're right that ESP32s with custom efuse SPI flash pin mappings (such as the PICO-D4 has) are not currently supported unless the stub is used. The SPI flash pins aren't configured probably by the bootloader.

It would be possible (but fiddly) for us to add support by having esptool write the relevant registers directly. However another option will be to detect this case and produce a clear error message asking to enable the stub.

Is there a reason you need `--no-stub` to work, or would the second option be acceptable for your use case?

ash-microconnect, 2019-09-12:
Thanks for updating this issue with the root cause. I'm glad to know that my hardware design was not responsible.

In general, I think having esptool print an error would be sufficient in most cases. My particular use case is somewhat unique (I'll describe it below) and I've found a way to resolve my issue anyway.

In my end product I have multiple ESP32-WROOM modules arranged "hub and spoke" style with a non-ESP32 micro as the hub.  The hub is the master to the slave ESPs. To facilitate firmware upgrade, I programmed the hub micro to use the esptool serial protocol to update each attached ESP32. To make it simpler, I used only the ROM loader commands; that way, I didn't have to have the hub micro first send the stub loader. This worked well whilst I was using the WROOM modules but now I need to replace one of them with a PICO-based module. To resolve the issue, I've modified the hub micro to first upload the stub and then continue the upgrade operation using the stub loader.

projectgus, 2019-09-12:
Hi @ash-microconnect ,

Sounds like an interesting product! Thanks for explaining.

Will leave this issue open until we add the error message for these cases.

Angus

negativekelvin, 2019-09-24:
`esptool.py --no-stub --chip=esp32 --port COM17 flash_id --spi-connection=6,17,8,11,16`

deng982217, 2019-10-17:
I meet a problem.  my chip is esp32-cam.
      项目使用了210066 字节，占用了(66)程序存储空间。最大为3145728 字节。

      全局变量使用了53552字节，(16)的动态内存，余留274128字节局部交量。最大为327680字节。

      esptool.py v2.6Serial port COM10Connect ing

      Chip is ESP32D0WDQ6 (revision 1)

      Features: wiFi, BT，Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme NonMAC: 3c: 71:bf:af:c5:3cUploading stub. . .Running stub

      出stub running.. .

      Tim Configuring flash size

      Warning: Could not auto-detect Flash size ( FlashID=0x0, sizeID=0x0) ，defaulting toCompressed 8192 bytes to 47...

      writing at 0x0000e000... (100 &)

      wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 580.0 kbit

      有吗A fatal error occurred: Timed out waiting for packet header

      Timed out waiting for packet header《

deng982217, 2019-10-18:
yesterday,In tried to ues another esp32-cam of my classmate and it truned out to be success.I think my hardware is broken.

tim-nordell-nimbelink, 2019-10-22:
Hi @ash-microconnect -

We have a non-ESP32 microcontroller, too, in our path.  Our solution was a bit different than yours - see 74ad6824e06976d4255607f356ad4093e38a43c6.  I don't know if that'll work for you, but the non-ESP32 microcontroller simply relays serial traffic through to the ESP32 in a pass through mode, and it only does the SYNC/baud rate switching portion of things.

- Tim

ash-microconnect, 2019-10-27:
Hi @tim-nordell-nimbelink 

Thanks for the input. Your suggestion wouldn't work for my exact case, as I'm doing an OTA-style update where the file is downloaded and stored locally before the central micro updates multiple ESPs in parallel.

gunarkroeger, 2020-01-07:
@projectgus I had the same problem for a similar situation of one pico trying to flash another. So the stub upload is needed for now..
What about the compression? even If -z is removed, the esptool still compresses the data before sending. Is there a way to send the data uncompressed to keep things simple?

projectgus, 2020-01-07:
@gunarkroeger you can pass either `-u` or `--no-compress` to flash without compression. (-z is the default if using stub, -u is the default if not using stub).

kowshik1234, 2020-07-23:
I wanna know what connections do I need to make if I am programming a custom board? I have Vcc, GND, Rx, Tx, EN and IO0 lines exposed.

radimkarnis, 2024-09-19:
This has been fixed in https://github.com/espressif/esptool/commit/bc9f2a656d3bc86097a7deb0c307327e2e62284f. Closing this ticket. Feel free to open a new one if you encounter a similar issue!

closed 2024-09-19 by radimkarnis