espressif/esptool #460: Flashing ESP8266 From MacBook over USB-C soft_reset does not work

telliottosceola, 2019-09-11:
* Operating system: Mac OS Mojave 10.14.5
* Python version: 2.7.10
* ESP hardware in use: ESP8266

# Full esptool.py command line as run: 
`esptool.py --after soft_reset --port /dev/tty.usbserial-DN05ESY1 write_flash 0x00000000 ~/Desktop/firmware.bin`

# Full output from esptool.py:
`esptool.py v2.3.1
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 342528 bytes to 244476...
Wrote 342528 bytes (244476 compressed) at 0x00000000 in 21.5 seconds (effective 127.3 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...`

# What is the expected behaviour?
Firmware is flashed, then user firmware begins running.

# Do you have any other information from investigating this?
I have found this works perfectly on a MacBook and iMac with native USB type B ports but will not work on computers with USB type C ports where an adapter is used.

# Is there any other information you can think of which will help us reproduce this problem?
The only way I know to reproduce this problem is using a Mac computer with USB type C ports.

So my main problem is that --after soft_reset does not appear to be working if I connect the ESP8266 to a computer via USB type C using an adapter.  Has anyone else reported an issue with the soft_reset while using USB type C?  This seems like a strange problem to have as I would assume since this is a soft reset the tool is sending data to the board to reset rather than hard_reset which uses the RTS line to reset the module.  Any ideas?

projectgus, 2019-09-16:
Hi @telliottosceola ,

Thanks for the report. Just to check: the hardware you're using is 100% the same except for the computer it's connected to, and the cable used to connect it?

If so, I can't explain this: the USB data protocol should be basically the same no matter what kind of plug is used, and the handshaking protocol is only different when a new connection is established.

Is it possible the ESP8266 is browning out because the adapter+cable combination can't supply enough current for the SDK & Wi-Fi to initialise?

It might be worth checking for other differentiating factors also: driver versions, macOS version, etc.

Angus

telliottosceola, 2019-09-16:
The flashing process succeeds every time, but It seems like for some reason on USB type C Mac computers the soft_reset simply does not work.  The reset button on the module has to be pressed.  This is not the end of the world for us, we can simply instruct technicians to press the reset button on the module after flashing is complete but I thought it was worth letting you all know about the issue.

I have checked Mac OS versions, Python versions, the USB to Serial driver(FTDI) and everything is identical between the two Macs(Soft reset works on one but not the other). The only difference I can find is one is a MacBook Air with USB type C and the other is an iMac with USB type B.  I have ensured use of the same USB cable and Power supply across testing as well of course using the same ESP8266 module.  We are flashing several hundred ESP8266 modules, this is the first time we have ran into this issue since we added another flashing computer(MacBook Air).

If there is any additional information I can provide please let me know.

prouast, 2020-10-01:
I am facing a similar issue with an ESP32:

- Flashing works well on my old MacBook with USB type B port.
- On my new MacBook with USB type C ports, I have tried and failed with both USB type B cable + Apple USB-C adapter and with an AmazonBasics USB Type-C to Micro-B 2.0 Cable

For both cables, flashing with USB type C port results in `A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header`.

@projectgus  @telliottosceola  I am new to programming with ESP32. Is there anything I can do to make it work with USB-C or will I have to resort to my old laptop with USB type B?

telliottosceola, 2020-10-01:
@prouast,

I gave up on using USB-C for programming Espressif modules.  I only work on Mac but luckily the iMac I'm on currently still has USB type B ports.  I hope by the time I'm forced to switch to a new computer with only USB-C ports this issue is resolved.  Sorry I couldn't be more help.

radimkarnis, 2026-02-11:
Revisiting this ticket after a few years, and I think it's safe to say this is not an issue anymore.

I tried reproducing this on MacOS Tahoe 26.2 - everything works as expected. What's more, the fact that there are no other similar reports confirms this.

Closing this. Feel free to open a new ticket if needed.