espressif/esptool #462: Detected flash size: Unknown, unable to write_flash

mkovero, 2019-09-23:
Hi! I'm unable to flash fresh ESP32 WROVER-B using esptool.py

* Operating system: Raspbian
* Python version: (`python -V` to check this) Python 3.7.3
* ESP hardware in use: ESP32D0WDQ5 (revision 1)

# Full esptool.py command line as run:

root@raspberrypi:/home/pi# esptool.py -c auto write_flash 0x1000 01.bin
esptool.py v2.7
Found 1 serial ports
Serial port /dev/ttyAMA0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:c6:38:78
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 753280 bytes to 423658...
Writing at 0x00005000... (7 %)
A fatal error occurred: Timed out waiting for packet header

root@raspberrypi:/home/pi# esptool.py flash_id
esptool.py v2.7
Found 1 serial ports
Serial port /dev/ttyAMA0
Connecting...
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:c6:38:78
Uploading stub...
Running stub...
Stub running...
Manufacturer: 00
Device: 0000
Detected flash size: Unknown
Hard resetting via RTS pin...

oot@raspberrypi:/home/pi# esptool.py erase_flash
esptool.py v2.7
Found 1 serial ports
Serial port /dev/ttyAMA0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:c6:38:78
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content

# What is the expected behaviour?
To be able perform write_flash or erase_flash.

# Do you have any other information from investigating this?
root@raspberrypi:/home/pi# espefuse.py -p /dev/ttyAMA0 summary
espefuse.py v2.7
Connecting.......
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
CLK8M_FREQ             8MHz clock freq override                          = 56 R/W (0x38)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1107 R/W (0x1)

Identity fuses:
MAC                    Factory MAC Address
  = 30:ae:a4:c6:38:78 (CRC 38 OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 1 R/W (0x1)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
root@raspberrypi:/home/pi# espefuse.py -p /dev/ttyAMA0 dump
espefuse.py v2.7
Connecting...
EFUSE block 0:
00000000 a4c63878 003830ae 0000a200 00000138 00000000 00000004
EFUSE block 1:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE block 2:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE block 3:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

This ESP32 is a part of new electrical design, so low voltage situation is possible due the fact we're on bit unfamiliar grounds, if you are able to identify whether this has been the case or do we have actual hardware problem with the flash, would be greatly appreciated.
Flash pins 11-16 are NC.

Thanks for your help!:)

mkovero, 2019-09-23:
SD0 and SD1 seem to have shorted, I believe might be related to this, no idea whether they should be though.

mkovero, 2019-09-23:
I have vague idea that I have possibly caused brown out in flash by using openocd default esp-WROVER configuration, which sets up 1.8V flash voltage, instead of required 3.3V with WROVER-B.
Possibility of brownout has been stated clearly in 
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

projectgus, 2019-11-27:
Hi @mkovero ,

Sorry I didn't get back to you earlier. Either of these explanations will cause flashing to fail with similar errors to what you describe.

Did you figure out the root cause?

Angus

mkovero, 2019-11-28:
Yeah issue was resolved, faulty hardware was the cause of the problem.
thanks anyways! :)

projectgus, 2019-11-28:
@mkovero Good to hear!

closed 2019-11-28 by projectgus