espressif/esptool #463: Fails at Uploading stub...

jdenver13, 2019-09-26:
(Please delete any lines which don't apply.)

* Operating system: Mac Mojave 10.14.6
* Python version: (`python -V` to check this) v2.7.10
* ESP hardware in use:
Arduino uno R3 with the chip removed powered by USB from Mac and the 3.3v and ground with Tx and Rx connected to the Sonoff.

# Full esptool.py command line as run:
esptool.py -p/dev/tty.usbmodem1443201 write_flash -fs 1MB -fm dout 0x0 sonoff_v6_6_0.bin

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.7
Serial port /dev/tty.usbmodem1443201
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:a7:ff:3f
Uploading stub...
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 4, in <module>
    __import__('pkg_resources').run_script('esptool==2.7', 'esptool.py')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 742, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 1667, in run_script
    exec(code, namespace, namespace)
  File "/Library/Python/2.7/site-packages/esptool-2.7-py2.7.egg/EGG-INFO/scripts/esptool.py", line 3152, in <module>
    _main()
  File "/Library/Python/2.7/site-packages/esptool-2.7-py2.7.egg/EGG-INFO/scripts/esptool.py", line 3145, in _main
    main()
  File "/Library/Python/2.7/site-packages/esptool-2.7-py2.7.egg/EGG-INFO/scripts/esptool.py", line 2860, in main
    esp = esp.run_stub()
  File "/Library/Python/2.7/site-packages/esptool-2.7-py2.7.egg/EGG-INFO/scripts/esptool.py", line 633, in run_stub
    self.mem_block(stub[field][from_offs:to_offs], seq)
  File "/Library/Python/2.7/site-packages/esptool-2.7-py2.7.egg/EGG-INFO/scripts/esptool.py", line 538, in mem_block
    self.checksum(data))
  File "/Library/Python/2.7/site-packages/esptool-2.7-py2.7.egg/EGG-INFO/scripts/esptool.py", line 369, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/Library/Python/2.7/site-packages/esptool-2.7-py2.7.egg/EGG-INFO/scripts/esptool.py", line 337, in command
    self.write(pkt)
  File "/Library/Python/2.7/site-packages/esptool-2.7-py2.7.egg/EGG-INFO/scripts/esptool.py", line 300, in write
    self._port.write(buf)
  File "build/bdist.macosx-10.14-intel/egg/serial/serialposix.py", line 552, in write
serial.serialutil.SerialTimeoutException: Write timeout
# What is the expected behaviour?
To flash the Sonoff_v6_6_0.bin file

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?
I tried reading the flash and got the same results at Uploading stub...

projectgus, 2019-10-18:
Hi @jdenver13 ,

This error "serial.serialutil.SerialTimeoutException: Write timeout" comes from the operating system trying to talk to the USB/serial adapter that's connected to Windows. It means "we tried to write to the port and the port never became available".

Usually this indicates a driver bug or some problem with the USB/serial adapter, there's nothing we can do to make it work from esptool itself.

Can you give any more details about the USB/serial adapter that you're using?

jdenver13, 2019-10-18:
Sorry I haven't updated this but your right, it was the USB/serial adapter sort of - I have a Mac book pro and the USB/serial adapter was plugged into a USB C Hub, it was that causing the problem. I thought as long as it established a connection all was fine with the hub and adaptor. I tried an Apple one and everything worked as expected. Thank you for your help. Regards

projectgus, 2019-11-27:
Hi @jdenver13 ,

Thanks for the update. That's weird, I am guessing it's something with the USB/serial driver in MacOS. Unfortunately not anything we can fix from the esptool side.

closed 2019-11-27 by projectgus