espressif/esptool #465: Reboot loop after esptool upload.....

delta010, 2019-10-01:
* Operating system: parrot os
* Python version: Python 2.7.16+
* ESP hardware in use: DOIT ESP32 DEVKIT V1 (ESP32)

# Full esptool.py command line as run: esptool --chip esp32 --port  /dev/ttyUSB1  write_flash -z 0x1000 xxxxx.bin 

# Full output from esptool.py : 
  esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 160MHz, VRef calibration in efuse, Coding Scheme None
MAC: 84:0d:8e:0b:bd:d4
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash...
Compressed 515872 bytes to 355877...
Took 1.87s to erase flash block
Wrote 515872 bytes (355877 compressed) at 0x00001000 in 34.9 seconds (effective 118.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

# Do you have any other information from investigating this?
 Arduino error code:
       
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x0f,d_drv:0x00,cs0_drv:0x01,hd_drv:0x00,wp_drv:0x04
mode:DOUT, clock div:2
load:0x40004b1c,len:1073751244
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
csum err:0xa4!=0x00
ets_main.c 371 
ets Jun  8 2016 00:22:57

Using Arduino IDE sketch upload and run without any problem but using esptool.py esp32 enter into a reboot loop

# Is there any other information you can think of which will help us reproduce this problem?
I Don't Know

comboner, 2019-10-05:
**I am having a very similar issue:**

I am running MacOS cataline Beta GM
Python version: 3.7.. (conda environment)
ESP hardware: ESP32D0WDQ6 (revision 1)

**esptool.py command I use is:**
esptool.py write_flash 0x1000 xxxx.bin

**full output from esptool.py:**

esptool.py v2.7
Found 4 serial ports
Serial port /dev/cu.usbserial-0001
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:94:5b:c1
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 747744 bytes to 420449...
Wrote 747744 bytes (420449 compressed) at 0x00001000 in 37.1 seconds (effective 161.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

**Serial output from Arduino serial monitor:**

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3f400020,len:157464
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3f400020,len:157464
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3f400020,len:157464
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3f400020,len:157464
ets Jun  8 2016 00:22:57
etc............

As delta010 I can verify, upload and run my code without issues with Arduino IDE.
As a next step I shall check the binaries using the verify_flash command.

Kind regards

comboner, 2019-10-06:
Sorry. Nevermind. I just had a look at the verbose output when Arduino verifies and uploads my code to esp. Apparently Arduino IDE compiles multiple files before flashing the binaries to esp. The Arduino IDE option "export compiled Binary" exports only one binary file. No matter at which offset I flash this file it does not work.

**Arduino IDE bin files with offset are:**
0xe000 .../Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/partitions/boot_app0.bin 
0x1000 ...Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/bin/bootloader_qio_80m.bin 
0x10000 /var/folders/_p/hdhpsrf108743bsn1mj8dy4h0000gn/T/arduino_build_334278/esp32-examples_HelloServer_HelloServer.ino.bin 
0x8000 /var/folders/_p/hdhpsrf108743bsn1mj8dy4h0000gn/T/arduino_build_334278/esp32-examples_HelloServer_HelloServer.ino.partitions.bin

projectgus, 2019-10-11:
Hi everyone,

I think you both have different issues.

@delta010 This looks like some kind of corruption when the ESP32 is loading the image from the flash. Can you try changing the flash mode from DOUT to DIO, and/or try using a slower flash clock speed. Both settings can be changed in Arduino, then re-compile and re-flash. If no combination of these changes work then the hardware might be faulty, more flash pins are used when booting compared to when flashing.

@comboner I am not familiar with ESP32 Arduino, but it looks like the "export compiled binary" features exports the app image only (`esp32-examples_HelloServer_HelloServer.ino.bin` in the list of binaries you posted). You can compare file sizes to confirm. The ROM boot log also supports the theory that the ESP32 is trying to boot an app image, not a bootloader image. This won't work. Try flashing all of the binaries listed to the offsets listed (you can do this in a single `write_flash` command.)

delta010, 2019-10-17:
Even I changed the flash mode it reboot after the sketch upload

projectgus, 2019-11-27:
Hi @delta010 ,

Are you still having this problem? Exactly what settings did you try? Is anything else connected to your DEVKIT board, or is it installed in a breadboard or anything else connected to the pins? If yes, try disconnecting the connections or removing the DEVKIT from the breadboard.

Angus

radimkarnis, 2020-12-09:
Hello everyone,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis