espressif/esptool #466: Invalid SLIP escape when reading flash

JulienDevillers, 2019-10-07:
(Please delete any lines which don't apply.)

* Operating system: ubuntu Ubuntu 19.04
* Python version: (`python -V` to check this) 2.7.16
* ESP hardware in use:  esp-wroom 4M

# Full esptool.py command line as run:
D=/home/test
$IDF_PATH/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 read_flash   0x20000 0x100000  $D/build/afficheurR.bin

# Full output from esptool.py (please copy and paste all lines of output)

esptool.py v2.8-dev
Serial port /dev/ttyUSB0
Connecting........___
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c4:4f:33:03:4e:4d
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
131072 (12 %)
A fatal error occurred: Invalid SLIP escape (0xdb, 0x0C)

# What is the expected behaviour?

no error

# Do you have any other information from investigating this?
The error does not occur at the same percentage.
esp-tools comes from esp-idf github. Commit hash : 46b12a560a29fa6ade07

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2019-10-10:
Hi @JulienDevillers,

Usually this kind of intermittent protocol error indicates some kind of problem with the serial link - data is being corrupted sometimes. Or it can be an issue with stable power to the module.

Could you post a photo of your hardware setup? How is power supplied to the module?

You can also try lowering the baud rate (to 115200 or less) and see if the connection becomes more reliable.

Angus

JulienDevillers, 2019-10-22:
Hi Angus,

After many tries, it appears that a computer's reboot solves this problem
that appears around 3 times a day....
I'm using a custom board. Oscilloscope shows nothing special with power
supply...

Thanks
Julien

projectgus, 2019-11-27:
Thanks for the update, @JulienDevillers . Closing as this seems like something with the hardware setup or the computer, not esptool.

closed 2019-11-27 by projectgus