espressif/esptool #469: A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

sothy, 2019-10-23:
(Please delete any lines which don't apply.)

* Operating system: ubuntu 16.04
* Python version: (`python -V` to check this): Python 3
* ESP hardware in use: Geekcreit® NodeMcu Lua ESP8266 ESP-12F WIFI Development Board 

# Full esptool.py command line as run:sudo python3 /root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py --port /dev/ttyS0 --baud 9600 read_mac

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.6
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

# What is the expected behaviour?
should return mac address

# Do you have any other information from investigating this?
I used miniterm to see what port device is connectioned.
$ sudo miniterm.py

--- Available ports:
---  1: /dev/ttyS0           ttyS0
--- Enter port index or full name: 1
--- Miniterm on /dev/ttyS0  9600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

sothy, 2019-10-26:
I myself made a mistake.  I found wrong port  to connect. Now I use the following tool to detect my device.
python -m serial.tools.list_ports
After that it works without any issues

pfeerick, 2019-10-28:
I take it the issue can be closed now then? :) Perhaps via the comment and close button? ;)

closed 2019-10-28 by sothy