espressif/esptool #470: how about a useful manual for noobs ?

Dakhaasje, 2019-10-26:
`Python 3.8.0 (tags/v3.8.0:fa919fd, Oct 14 2019, 19:37:50) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> $ pip install –upgrade esptool
SyntaxError: invalid syntax
>>> python -m pip install esptool
SyntaxError: invalid syntax
>>> pip install –upgrade esptool
SyntaxError: invalid syntax
>>> pip2 install esptool
SyntaxError: invalid syntax
>>> python setup.py install
SyntaxError: invalid syntax
>>> python -m pip install esptool
SyntaxError: invalid syntax
>>> `
I just need esptool to flash(wipe) my 'not responding' sonoff pow r2.
 and python to run the tasmota tool. 
I'm using w10 on amd x64.

pfeerick, 2019-10-28:
? You want a manual on how to use python? Instead of running python and then entering a command that should be entered at the command prompt... at your command prompt, run `python -m pip install esptool`.

That's `python`, told to load/run the pip module `-m pip`, with the `install esptool` parameters. 

Why not use the NodeMCU PyFlasher [as recommended by the Tasmota folks](https://github.com/arendst/Tasmota/wiki/Flashing)? Then you have easy to use GUI which will do exactly what you want.

Dakhaasje, 2019-10-28:
because that only went by  in a flash in a dos-window. not the kind of flash I was looking for.

pfeerick, 2019-10-28:
> not the kind of flash I was looking for.

Nice one ;) 

Next time, if you're doing command line stuff... open the the Command Prompt / CMD first... then the window won't disappear on you ;) Glad you got it working. You right to close the issue now?

Dakhaasje, 2019-11-07:
problem solved. With the new ui I was mistaken.

closed 2019-11-07 by Dakhaasje