espressif/esptool #471: Sonoff T1 2ch  fatalerror time out

hassan4mz, 2019-10-28:
i am trying to flash my sonoff t1  to tasmota   and i started  to fowling the website 
https://blog.hobbytronics.pk/sonoff-original-firmware-backup-restore/
to take back up  first  the problem comes on the erase_flash  step  

C:\esp2>esptool.py --port COM4 erase_flash
esptool.py v2.9-dev
Serial port COM4
Connecting........_____....._
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 60:01:94:99:8c:e0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content

after that the key not working at all  please help me 
it working only with sonoff basic fine but on sonoff t1 not working i have 3  same issues

projectgus, 2019-10-31:
Hi @hassan4mz,

It's possible that the erase took longer than expected, but succeeded even though there was a timeout.

If you go to the next step and flash the Sonoff, does it work?

Angus

hassan4mz, 2019-11-01:
i trying a lot of time but not working the tasmota firmware is  100 % uploading but not working the led not blinking and the button not responding  
and when i try to restore the  backup firmware not become 100%  uploading a lot of  timeout same this massage (A fatal error occurred: Timed out waiting for packet content)

hassan4mz, 2019-11-05:
ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld

projectgus, 2019-11-27:
Hi @hassan4mz,

If esptool is successfully uploading the firmware with no errors then this is probably something with the Tasmota binary or the flashing settings (maybe try `write_flash --flash_mode dio` for example).

Otherwise I'm afraid you'll need to ask the Tasmota people for assistance, esptool is just responsible for writing the firmware not for the contents of the firmware.

closed 2019-11-27 by projectgus

Resurrection2981, 2021-05-07:
> 我尝试了很多时间但无法正常工作，但 tasmota 固件已 100％上传但未工作，导致 LED 不闪烁且按钮未响应
> ，当我尝试恢复备份固件时未变为 100％上传很多超时，这与这种情况相同（发生致命错误：等待数据包内容超时）

Try to pullup gpio15 when "Erasing flash (this may take a while)..." show up.