espressif/esptool #478: esp8266 (ch340) - read flash error on linux but windows ok.

tomudo, 2019-11-09:
(Please delete any lines which don't apply.)

* Operating system: linux /window
* Python version: python 3.7
* ESP hardware in use: ESP8266 (clone - ch340)

# Full esptool.py command line as run: 
esptool.py --port /dev/ttyUSB0 --baud 57600 --before default_reset --after hard_reset read_flash 0x00000 0x400000 backup.img

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: xxxxxxxxx
Uploading stub...
Running stub...
Stub running...
61440 (1 %)
A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xff8 bytes

# What is the expected behaviour?
can read flash 100% without error

# Do you have any other information from investigating this?
I install esptool in Windows python 3.7 by pip that work perfectly

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2019-11-12:
Hi @tomudo ,

If everything else is 100% the same (same modules, same USB/serial converter, same cables, same power source, etc) then the difference is probably related to the serial drivers.

If you try some different baud rates (9600, 115200, 230400) is there any difference?

Is there any update available to the CH340 driver on Windows?

tomudo, 2019-11-12:
> 
> 
> Hi @tomudo ,
> 
> If everything else is 100% the same (same modules, same USB/serial converter, same cables, same power source, etc) then the difference is probably related to the serial drivers.
> 
> If you try some different baud rates (9600, 115200, 230400) is there any difference?

Baud rates 9600 on Linux (Ubuntu 18.04) really slow and sometimes works but most not. More than 9600 that not work.

Less than 57600 on windows 10 works. More than 57600 does not works.

It looks like depend on driver.

SUNsung, 2020-04-29:
### ESP32, linux

**same problem**
previously successfully used **esptool** for similar purposes
I even have *.sh for such purposes (creating a snapshot or recording from a snapshot)
About six months I did not use **esptool** for such purposes, but today it took

or hangs at 1%
or in the process gives an error:
`A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xf84 bytes`

**0xf84** may change but **0x1000** invariably

I tried and change the speed and change the markup and completely clean and compile again and then do a snapshot
no changes

I think this is a problem on your part
this problem was not previously

I will try to find an archive hard drive on which I have a dump of a system with a stable build
I’ll complement the thread by the result

SUNsung, 2020-04-29:
yes the problem is in the new version
build 2019 esptool  has no given problem

SUNsung, 2020-09-23:
I think the ticket can be closed.
The error takes place, but it has dependencies.

if you get this error:
1 - install an earlier version of esptool
2 - use another USB-to-TTL adapter
3 - update the firmware of your USB-to-TTL adapter

I have many different USB-to-TTL and they all either depend on the esptool version or on the firmware of the adapter itself (there are a couple of pieces on FTDI and the chip with the old firmware causes a similar response, but if you update to the official one, then everything is OK)

radimkarnis, 2020-12-09:
Hello @SUNsung,
I’m closing this issue because you have proposed to do so and it has been inactive for a few months. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis