espressif/esptool #479: esptool.py command not found

jakudo, 2019-11-09:
Hi,

I've tried to follow help in issue #427, however, didn't manage it to work. 
After entering pip show -f esptool I get the address as:
.../.../.../bin/esptool.py

But I can't figure out what to do with that.
I have the same on Mac OS X Catalina and also in Ubuntu 19.04

[image]

Any idea? Thanks a lot.

projectgus, 2019-11-14:
Hi @jakudo ,

You have two options. One is to run esptool.py as `.../.../.../bin/esptool.py ...` directly, this will work but it's a bit kludgy.

The other is to add the `/bin` directory with Python scripts to your PATH. Find the absolute full path name for this `.../.../.../bin` directory and then add this directory to your PATH as shown here: https://www.architectryan.com/2012/10/02/add-to-the-path-on-mac-os-x-mountain-lion/

jakudo, 2019-11-14:
Hi,

thanks a lot for reply. I will try that.
However, for some unknown reason, the esptool.py started to work on Ubuntu after I restarted the PC, so happily it works at least on one of my computers.
Tried to restart also iMac, but still the same, so just for learning purpose, I will try that there:)

Thanks!

closed 2019-11-14 by jakudo