espressif/esptool #480: Error flashing ESP8266-ESP01 - Invalid head of packet (0x00)

lmbautista, 2019-11-11:
* Operating system: macOS Mojave 10.14.5
* Python version: Python 2.7.15
* ESP hardware in use: ESP8266 ESP-01

# Full esptool.py command line as run:
```
esptool.py --trace -p /dev/cu.wchusbserial1420 --baud 115200 write_flash 0x00000 v0.9.2.2\ AT\ Firmware.bin -u
```

# Full output from esptool.py (please copy and paste all lines of output)
```
esptool.py v2.8
Serial port /dev/cu.wchusbserial1420
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.103 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.106 Timed out waiting for packet header
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.103 Timed out waiting for packet header
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 31 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 01080200070712   | ... ...........
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.002 Read 1 bytes: 20
TRACE +0.001 Read 31 bytes: 
    0000c0c001080200 070712200000c0c0 | ........... ....
    0108020007071220 0000c0c0010802   | ....... .......
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.001 Read 32 bytes: 
    00070712200000c0 c001080200070712 | .... ...........
    200000c0c0010802 00070712200000c0 |  ........... ...
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.001 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=78000060
TRACE +0.000 Write 14 bytes: c0000a04000000000078000060c0
TRACE +1.015 Read 1 bytes: 00
TRACE +0.000 Read invalid data: 00
TRACE +0.000 Remaining data in serial buffer: 
    0000000000000000 0000000000000000 | ................
    0000000000000000 00000000000000   | ...............
```
# What is the expected behavior?
To finish successfully the flash process.

# Do you have any other information from investigating this?
I've tried the following changes:
- Modify baud rate to 9600: even worse as It does not connect with ESP.
- Review of connections several times. I could attach images if needed.
- 

# Is there any other information you can think of which will help us reproduce this problem?
I modified the baud rate from
`AT+UART_DEF=9600,8,1,0,0`
to
`AT+UART_DEF=9600,8,1,0,3`

and then the AT stop working.

After trying to revert it unsuccessfully I thought about flash it again (as I did several times in the past) but now I got this error.

BTW: I could attach any additional information required. 

Thanks in advance.

projectgus, 2019-11-12:
Hi @lmbautista ,

The log seems to indicate that the ESP8266 starts into reset mode and responds to the initial "sync" request, and then sends a garbage response of all zeroes immediately after.

How are you powering the ESP8266? Power glitches can cause weird failures modes like this.

AT commands won't change what esptool does as it resets the ESP8266 into a special bootloader mode, separate from any firmware.

lmbautista, 2019-11-23:
Thanks for the help @projectgus. Despite I tried several tutorials, steps and ways to tackle the problem seems I'm not able to flash the ESP8266 again so I'll buy another one 😅 

Once again thanks

closed 2019-11-23 by lmbautista