espressif/esptool #481: Does not work correctly with cp2102 in Linux mint 19.1

AlexanderSemenov1, 2019-11-12:
Does not work correctly with cp2102 in Linux mint 19.1
Linux alex-pc 5.3.0-19-generic #20~18.04.2-Ubuntu SMP Tue Oct 22 18:09:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
__________________________________________________________________________

alex@alex-pc:~/file-bin$ esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0 esp8266-20190529-v1.11.bin
esptool.py v2.7
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:5a:74:c6
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 617880 bytes to 402086...
Wrote 617880 bytes (402086 compressed) at 0x00000000 in 35.5 seconds (effective 139.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
_________________________________________________________________________

The Putty program sends garbage to the terminal.
  I tried to set different speeds, and nothing helped. It seems that the firmware is not correctly written.
This problem is only on devices with cp2102 !. ch340 works great!

projectgus, 2019-11-12:
@AlexanderSemenov1 Are there different development boards? I think most likely there is some difference in onboard flash config, rather than the serial chip being significant.

If you run `esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fm dio -ff 40m 0 esp8266-20190529-v1.11.bin` instead, does it work?

(The difference to the original command is the `-fm dio -ff 40m` arguments to change the flash mode and speed.)

If that doesn't work, what output do you get from PuTTY (or similar) at baud rate 74880?

AlexanderSemenov1, 2019-11-13:
Thank you for your help !!!.
This command helped me and the firmware was installed as it should.
Before, I never explicitly specified the -fm or -fs options and everything was written as it should. It doesn't matter esp8266 or esp32 b it doesn't matter which converter was installed cp2102 or ch340.
Now you need to explicitly specify the -fm parameter on the board with cp2102, and on the board with ch340 everything is written correctly.
It seems that micropython developers no longer specify these parameters in the firmware header.

projectgus, 2019-11-14:
Hi @AlexanderSemenov1 ,

Glad it's working now.

It's possible that MicroPython has changed the default from DIO to QIO to get more flash performance (QIO mode is approximately twice as fast, but not all flash chips support it)..

To be clear, the difference isn't CP2102 vs CH340, this is just the USB to serial adapter chip. It happens that one development board which has CP2102 adapter and an ESP8266 connected to a DIO-only flash chip. The other development board has CH340 adapter and an ESP8266 connected to a flash chip that supports QIO mode. 

With different hardware, you could easily come across a different combination again.

closed 2019-11-14 by projectgus