espressif/esptool #483: esptool.FatalError: Failed to connect to ESP8266: Time out waiting for packet header

Jeongmin-26, 2019-11-16:
Hello.

I'm using Arduino LOLIN (WEMOS) D1/ESP8266.
There is a problem with ESP8266 connection, so I keep getting an error message. The situation is the same with a simple example.
error message : esptool.FatalError: Failed to connect to ESP8266: Time out waiting for packet header
Please, I need your help :(

* Operating system: Window10 x64
* Python version: (`python -V` to check this) 2.7.17
* ESP hardware in use: ESP8266

# Full esptool.py command line as run: 2.8

# Full output from esptool.py (please copy and paste all lines of output)

# What is the expected behaviour?

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2019-11-19:
@Jeongmin-26 Can you please supply the full esptool.py command line and the full output (all lines), as asked for in the issue template?

projectgus, 2019-11-27:
Closing due to lack of required information for debugging.

closed 2019-11-27 by projectgus