espressif/esptool #484: Wiki Issue

sastorer-ms, 2019-11-19:
"Here is a sample extract, showing a READ_REG command and response:

TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=1400f43f
TRACE +0.000 Write 14 bytes: c0000a0400000000001400f43fc0
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200620100000000c0
TRACE +0.000 Received full packet: 010a0200620100000000"

Can you explain the "Read 1 bytes" line? Is this a typo? It does not follow the SLIP protocol for reading 1 byte...

projectgus, 2019-11-19:
Hi @sastorer-ms ,

Thanks for pointing this out, the trace is correct but there is some layering going on which is not immediately evident from reading the trace outputs.

I've just added the following explanatory text to the wiki:

> Note that multiple protocol layers are represented in the logs. The "Write X bytes" lines show exactly which bytes are being sent "over the wire", including SLIP framing. Similarly the "Read X bytes" lines show what bytes are being read over the wire, including any SLIP framing. Once a full SLIP packet is read, the same bytes - as a SLIP payload with any escaping removed - appear in the "Received full packet" log lines.

Does that explain the behaviour adequately?

Angus

sastorer-ms, 2019-11-19:
Yep, thanks for the clarification. Can you add in any explanation as to why the layering is happening? Why didn't the whole SLIP packet get read at once?

projectgus, 2019-11-20:
> Why didn't the whole SLIP packet get read at once?

We have a asynchronous serial link, so bytes are sent over the wire one at a time and all we ever get is X bytes at a time. What we read in a single read() call could be the contents of 0, 1, or more complete SLIP packets.

BTW, the esptool implementation has an implementation-specific optimisation where it reads one byte first (with a timeout), then tries to read the number of remaining bytes reported as waiting in the operating system RX buffer. This is the most efficient way to read an unknown number of bytes using pyserial APIs, as if you supply a large buffer and a timeout then it will keeping trying to fill the entire buffer until the timeout expires: https://github.com/pyserial/pyserial/blob/master/serial/serialposix.py#L538

But even without this optimisation, there's no guarantee that a single read() call will capture exactly one SLIP packet. (The wiki shows an example of syncing where 4.5 response packets are received in a single read.)

> An explanation of where "wait_response=1 timeout=3.000" come from is also missing

Some of this stuff is probably better understood by reading the esptool source code, the "--trace" function is mostly intended as a way to look at the serial protocol itself (or debug interactions with bad hardware) rather than a way to understand the full workings of esptool.

* https://github.com/espressif/esptool/blob/master/esptool.py#L68
* https://github.com/espressif/esptool/blob/master/esptool.py#L334

sastorer-ms, 2019-11-20:
Thank you so much for your time and help.

closed 2019-11-20 by sastorer-ms