espressif/esptool #485: A fatal error occurred: Invalid head of packet (0x46)

HashemDekimo, 2019-11-20:
(Please delete any lines which don't apply.)

* Operating system: windows 8.1
* Python version: 3.7
* ESP hardware in use: ESP32-LyraTD-MSC

# Full esptool.py command line as run:
esptool.py dump_mem 0x8000 65536 iram0.bin
# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.8
Found 1 serial ports
Serial port COM4
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme
None
Crystal is 40MHz
MAC: 30:ae:a4:c3:88:50
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Invalid head of packet (0x46)

# What is the expected behaviour?
not to get this error
# Do you have any other information from investigating this?
no
# Is there any other information you can think of which will help us reproduce this problem?
no

closed 2019-11-20 by HashemDekimo

jbrepogmailcom, 2020-04-18:
Hello you have opened this issue and then you closed it. Does it mean you resolved it? How please?

HashemDekimo, 2020-04-22:
> .[image: Web Bug from
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url": "

Hi Espressif/Esptool,

That was long time ago, all what i remember , i didn't do any modifications
, just configuration change.

sorry if i wasn't very helpful.

regards.