espressif/esptool #488: Can't upload to SPIFFS with 2.6.2

kfeger, 2019-12-01:
I try to upload to SPIFFS (upload to Flash works fine).
----------------------------- Delete below -----------------------------

Most failures to connect, flash, etc. are problems with the hardware.

Please check any guide that came with your hardware, and also check these troubleshooting steps:

https://github.com/espressif/esptool/#troubleshooting

If still experiencing the issue, please delete the text in this box and then provide as many details as possible below about your hardware and computer setup.

----------------------------- Delete above -----------------------------

# Operating system

# Python version

Can run `python -V` to check this.

# What Chip

(For example: ESP8266, ESP32-PICO-D4, ESP32-WROOM32 module, etc)

# What development board or other hardware is the chip attached to

(For example: DevKitC, NodeMCU board, plain module on breadboard, etc)

If your hardware is custom or unusual, please attach a photo to the issue.

# Is anything else attached to the development board, except for the serial flasher connections?

Example: GPIO 18 & 19 are connected to I2C devices.

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

Example: No IDE, Windows command prompt

Example 2: Arduino ESP8266 IDE version 2.5.77

# Full esptool.py command line that was run:

esptool.py -p COM999 write_flash 0x0 bootloader.bin

# Full output from esptool.py

(Please copy and paste all lines of output here)

# Do you have any other information from investigating this?

Example: The command succeeds 1 in every 3 tries

# Is there any other information you can think of which will help us reproduce this problem?

Example: I also have a Windows 95 PC, esptool command works correctly on this PC.

kfeger, 2019-12-01:
2.6.1 gives the same error. 2.6.0 works fine. Here the output from 2.6.0
****************************
[SPIFFS] data    : C:\Users\Besitzer\Documents\Arduino18-xeon\ESP_ISS\data
[SPIFFS] size    : 192
[SPIFFS] page    : 256
[SPIFFS] block   : 4096
/index.html

/ISS150.png

/settings.html

/setup.png

[SPIFFS] upload  : C:\Users\Besitzer\AppData\Local\Temp\arduino_build_372128/ESP_ISS.spiffs.bin
[SPIFFS] address  : 0xCB000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : COM6
[SPIFFS] speed    : 115200
[SPIFFS] python   : python.exe
[SPIFFS] uploader : C:\Users\Besitzer\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.0\tools\upload.py

esptool.py v2.7

Serial port COM6

Connecting....

Chip is ESP8266EX

Features: WiFi

Crystal is 26MHz

MAC: 18:fe:34:dc:ff:42

Uploading stub...

Running stub...

Stub running...

Configuring flash size...

Auto-detected Flash size: 1MB

Compressed 196608 bytes to 60539...

Writing at 0x000cb000... (25 %)
Writing at 0x000cf000... (50 %)
Writing at 0x000d3000... (75 %)
Writing at 0x000d7000... (100 %)
Wrote 196608 bytes (60539 compressed) at 0x000cb000 in 5.3 seconds (effective 294.0 kbit/s)...

Hash of data verified.

projectgus, 2019-12-02:
Hi @kfeger ,

```
esptool: error: unrecognized arguments: --end
```

The problem is the command line that Arduino IDE is building when it runs esptool.py. There isn't anything that can be done about it from the esptool.py side, the Arduino IDE has to send a valid command.

This issue suggests that if you update the Upload plugin in Arduino then it will fix it: https://github.com/esp8266/Arduino/issues/6864

closed 2019-12-02 by projectgus

kfeger, 2019-12-02:
Hi Angus,

I downloaded the new Upload plugin. But since I’m by no means a Java person: How do I come from the supplied ESP8266FS.java tot the needed .jar file? Make.sh will not work under Win10…

 

This issue suggests that if you update the Upload plugin in Arduino then it will fix it: esp8266/Arduino#6864 <https://github.com/esp8266/Arduino/issues/6864>

kfeger, 2019-12-02:
Sorry to disturb again but: Got it. Version 0.5 works fine.

Karl