espressif/esptool #489: ESP32 - TTGO - read_flash fail -  ESP32 ROM does not support function read_flash

ds2k5, 2019-12-06:
# Operating system
Debian Linux

# Python version
Python 2.7.17
Python 3.7.5

# What Chip
ESP32D0WDQ6

# What development board or other hardware is the chip attached to
TTGO

# Is anything else attached to the development board, except for the serial flasher connections?

USB

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

no

# Full esptool.py command line that was run:

# esptool -t --chip esp32 -p /dev/ttyUSB0 -b 115200 read_flash 0x0 4194304 esp32_TTGO.bin
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.007 Read 1 bytes: 65
TRACE +0.000 Read invalid data: 65
TRACE +0.000 Remaining data in serial buffer: 
    7473204a756e2020 3820323031362030 | ts Jun  8 2016 0
    303a32323a35370d 0a0d0a7273743a30 | 0:22:57....rst:0
    78312028504f5745 524f4e5f52455345 | x1 (POWERON_RESE
    54292c626f6f743a 3078332028444f57 | T),boot:0x3 (DOW
    4e4c4f41445f424f 4f54285541525430 | NLOAD_BOOT(UART0
    2f55415254312f53 44494f5f5245495f | /UART1/SDIO_REI_
    52454f5f56322929 0d0a77616974696e | REO_V2))..waitin
    6720666f7220646f 776e6c6f61640d0a | g for download..
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 111 bytes: 
    0108040012205555 00000000c0c00108 | ..... UU........
    0400122055550000 0000c0c001080400 | ... UU..........
    1220555500000000 c0c0010804001220 | . UU........... 
    555500000000c0c0 0108040012205555 | UU........... UU
    00000000c0c00108 0400122055550000 | ........... UU..
    0000c0c001080400 1220555500000000 | ......... UU....
[... 167 lines cut ...]
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 24:6f:28:FF:FF:FF
Hard resetting via RTS pin...

# esptool flash_id
esptool.py v2.6
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting.......
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:FF:FF:FF
Enabling default SPI flash mode...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

# Do you have any other information from investigating this?

always

# Is there any other information you can think of which will help us reproduce this problem?
no

ds2k5, 2019-12-08:
with Version 2.8 of esptool it works!

closed 2019-12-08 by ds2k5

pkoevesdi, 2021-01-19:
Cannot confirm that it works with esptool 2.8 (`sudo apt install esptool` on ubuntu 20.04), but can confirm it working with v3.1-dev, taken from esp-idf/components/esptool_py/esptool/esptool.py (ESP-IDF 4.3)

crasu, 2021-01-30:
Works with 3.0 as well