espressif/esptool #490: Fatal Error Failed to connect to ESP8266: Timed out waiting for packet header

imaNNeoFighT, 2019-12-07:
Hello, 
here is my bord 

[Geekcreit® UNO + WiFi R3 ATmega328P+ESP8266 Module mémoire 32 Mo USB-TTL CH340G pour Arduino unités](https://fr.banggood.com/Geekcreit-UNOWiFi-R3-ATmega328PESP8266-32Mb-Memory-USB-TTL-CH340G-Module-For-Arduino-Uno-NodeMCU-ESP8266-p-1205444.html?gmcCountry=FR&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=frank&utm_campaign=ssc-frg-summersale-all-usw&utm_design=152&ad_id=359008578876&gclid=CjwKCAiAuK3vBRBOEiwA1IMhusYFAiE7Z3QlmuoLjwhne9vHXt7WSmheKK9EoJwNDsWdwoViou3l8BoCFkgQAvD_BwE&cur_warehouse=CN)

<img src="https://user-images.githubusercontent.com/7009300/70379804-c4e57a80-1946-11ea-953c-ffedb903a5e9.png"
    width=300 />

When I try to upload a sketch, I get this error:
```
esptool.py v2.8
Serial port /dev/cu.wchusbserial1410
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/Users/iman/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/Users/iman/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/Users/iman/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
```

[see full log](https://justpaste.it/5d46q)

projectgus, 2019-12-08:
Hi @imaNNeoFighT ,

I'm not familiar with this board but please check you have the DIP switches set for the ESP8266 to go into firmware upload mode. This board doesn't do it automatically:

[image]

(This is from the [English version of the product page](https://www.banggood.com/Geekcreit-UNOWiFi-R3-ATmega328PESP8266-32Mb-Memory-USB-TTL-CH340G-Module-For-Arduino-Uno-NodeMCU-ESP8266-p-1205444.html) but the French version has the same table.)

After changing the DIP switches you will need to unplug and replug the board, or press the "ESP" button which I'm guessing is a reset button for the ESP chip.

imaNNeoFighT, 2019-12-09:
I didn't now that.
Worked like a charm.
Thanks!

closed 2019-12-09 by imaNNeoFighT

mmors01, 2020-02-27:
Hi! how to solve this problem? thank you :)
I'm using Arduino to program ESP8266 - ESP-01 wifi module

An error occurred while uploading the sketch
esptool.py v2.8
Serial port COM8
Connecting......................................___
Traceback (most recent call last):
File "C:\Users\acer\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in 
esptool.main(cmdline)
File "C:/Users/acer/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2890, in main
esp.connect(args.before)
File "C:/Users/acer/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 483, in connect
raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

Mwalima, 2020-03-02:
Hi I am having the same problem

paco9595, 2020-03-19:
hi, I'm having the same problem

neilsroberts, 2020-03-26:
I'm experiencing this too. Furthermore it only started occuring whith a new pc this january, so I'm suspicious it is Windows related (W10-1909), as I've not changed any of my other hardware, and it has reprogrammed a couple of times, but then loses the ESP on the COM port. Often then making the connect/disconnect noise.

esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM4
Traceback (most recent call last):
  File "C:\Users\NR\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/NR/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2652, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:/Users/NR/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 222, in __init__
    self._port = serial.serial_for_url(port)
  File "C:/Users/NR/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/pyserial\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:/Users/NR/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/pyserial\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)
serial.serialutil.SerialException: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)

ArianaFlores20, 2020-03-26:
hi, I'm having the same problem :(
 
esptool.py v2.8
Serial port COM7
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\lenovo\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/lenovo/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/lenovo/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

chuckyro, 2020-03-29:
Hi all,

yes, there is a problem with the new esptool.py script responsible for uploading sketchs to arduino boards. (in my case esp12). if you downgrade the library for esp8266 to ver 2.6.0., probably it will work.

try this case!

we need to inform the dev guys (on github) to patch / change this script.

neilsroberts, 2020-03-30:
I've had the same problem using esptool.py v2.7, 2.6.and 2.8. I've just wired in the Reset and DTR lines on the FTDI board to ESP-01, they were not needed before, but I'll try anything, and I continue to get same output.

vigiles, 2020-04-01:
The same problem. Mac, ArduinoIDE1.8.12,ESP8266-01
`esptool.py v2.8
Serial port /dev/cu.usbmodem14301
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/Users/hi/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/Users/hi/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/Users/hi/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header`
upload fail.so ,should i use 2.5.x module ?

neilsroberts, 2020-04-03:
Minor success, yesterday I detached everything from the laptop, booted, plugged in the serial connector and ESP8266-01S, and not only updated the firmware loaded a sketch as well. On 2 seperate 01S boards. Secret is to release the ground connection to GPIO0 soon after the triple blue flash at ESP boot. previously I'd left it grounded as I had many times before.

So, thought I'd do the same again today with a ESP8266-01, reflashed the firmware to v3.0.3 with everything connected on docking station, but the sketch wouldn't load with IDE, same issue as before the esptool is failing to connect. Tried rebooting pc off dock and got the same result, interestingly the firmware flasher wouldn't connect either. So I've no idea what's going on.

Windows 10
esptool is v2.8 (tried 2.5, 2.6, 2.7)
IDE 1.8.12
Espressif firmware tool v3.6.8

chuckyro, 2020-04-03:
go to boards manager ... find esp8266 board .. select version 2.5.2 and install it (downgrade it)

neilsroberts, 2020-04-03:
which bit of tried  2.5, 2.6, 2.7 was not undestood? Anyway also tried 2.5.2; it worked - once, then I get same problem. Upgraded to latest and it worked once again before going to same problem.

vigiles, 2020-04-04:
> which bit of tried 2.5, 2.6, 2.7 was not undestood? Anyway also tried 2.5.2; it worked - once, then I get same problem. Upgraded to latest and it worked once again before going to same problem.

2.5.2 still fail

vigiles, 2020-04-04:
> which bit of tried 2.5, 2.6, 2.7 was not undestood? Anyway also tried 2.5.2; it worked - once, then I get same problem. Upgraded to latest and it worked once again before going to same problem.

I found sth. I used 2.5.2 or 2.6.3 all success. 
ArduinoUNO board connect esp8266-01, 3.3v <--> VCC, GND <--> GND, TX <--> RX, RX <--> TX. OK, go  on.
**First, the wirings of esp8266's EN(CH_PD) and IO0 pin is disconnect.**
Next, ArduinoIDE upload your code, 
When console print "Connecting........_____...", 
    **First! connect IO0 and GND pins, then, connect EN and 3.3v pins.**
You will see sth in console like this:
    Chip is ESP8266EX
    Features: WiFi
    Crystal is 26MHz
    MAC: xx:xx:xx:xx:xx:xx
    Uploading stub...
    Running stub...
    Stub running...
    Configuring flash size...
    Auto-detected Flash size: 1MB
    Compressed 313968 bytes to 225259...
    Writing at 0x00000000... (7 %)
After a while, it will be complated. You should disconnect IO0 and EN, then just connect EN again. esp8266 will be working.

vigiles, 2020-04-04:
> which bit of tried 2.5, 2.6, 2.7 was not undestood? Anyway also tried 2.5.2; it worked - once, then I get same problem. Upgraded to latest and it worked once again before going to same problem.

I found sth. I used 2.5.2 or 2.6.3 all success.
ArduinoUNO board connect esp8266-01, 3.3v <--> VCC, GND <--> GND, TX <--> RX, RX <--> TX. OK, go on.
**First, the wirings of esp8266's EN(CH_PD) and IO0 pin is disconnect.**
Next, ArduinoIDE upload your code,
When console print "Connecting........_____...",
**First! connect IO0 to arduino's GND pins, then, connect EN to arduino's 3.3v pins.**
You will see sth in console like this:
_Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 313968 bytes to 225259...
Writing at 0x00000000... (7 %)_
After a while, it will be completed and success. **You should disconnect IO0 and EN, then just only connect EN again.** esp8266 will be working.

neilsroberts, 2020-04-06:
Yes, it still failed, it worked once, I removed the ESP package in Arduino15 folder, loaded 2.5.2 (as I did with other packages also), it worked for 1 time, then I get same fail message, same as when I upgraded to other versions. It sometimes works once, then fails. It's not only frustrating, but really odd the way it ehaves.

vigiles, 2020-04-06:
> Yes, it still failed, it worked once, I removed the ESP package in Arduino15 folder, loaded 2.5.2 (as I did with other packages also), it worked for 1 time, then I get same fail message, same as [...]

Yes. I made it yesterday on 2.6.3 version, the computer keep working, and arduino uno keep connected with esp8266. But I closed the Mac when I went to bed and disconnected the wires, it's fail again all day now.

neilsroberts, 2020-04-13:
Finally discovered my problem yesterday, a dry joint in the connection of my switch to ground, so when I held it a certain way, it worked, else it didn't.

Frustrating as I'd checked the switch, but when checking ground to GPIO I duscovered it wasn't being pulled down. Fixed the joint, IDE 1.8.12 and ESP 2.6.3 works after reboot and multiple times.

Lesson to self, check the circuit not the component!

sankharr, 2020-05-17:
I got the same problem in my nodemcu esp8266. That was because i have plugged a wire to D4 pin while the code was uploading. once i removed it, the code got uploaded successfully

Andy8266, 2020-06-07:
I have the same problem. I've tried switching cables. I've tried hosting the ESP8266-01 from Arduino Uno and Nano. I installed a complete new Arduino IDE on another PC.  I've installed the CN340 USB serial drivers. I'm holding down GPI0 during reset, then releasing it. I have an external 3.3V power supply to get enough juice. I pulled enable and reset high with 10K resistors.  All of this changed nothing. The espytool.py reports "timed out waiting for packet header".  I'm using the generic8266 board drivers. I've tried grounding and not grounding the RST pin on the Arduino. I've used ESP8266 modules from two different suppliers, though I expect they use the same source, as they look identical. Help!  I've spent days trying to figure this out. The blue light blinks on the ESP8266. It works fine in AT mode.

neilsroberts, 2020-06-08:
Do the LEDs flash in same sequence when you power up with/without GPI0 pulled to ground? If they do the ESP is not going into flash mode and you get this message. It is then a case of determining why your ground link is not working. Test the switch, joints, etc with a multimeter, that is how I found my dry joint.

Andy8266, 2020-06-08:
Thank you Neil for responding.  I used AT mode to reset the UART default to 74,880 in order to see the full boot-up message on the Arduino IDE serial monitor.  Boot mode(1,6) was correct for downloading. I found that I could upload if I kept this baud rate in place of the 115,200 default AND waited about 15 seconds after a reset for the download mode to get itself sorted out.  It is unfortunate that the built-in LED on my unit is at D1 and not D2 as most of the references suggest. I found that tip after some searching. With all that I'm finally up and running!  Oh ... and unlike the Arduino code, it doesn't start immediately after upload, you need to reset the ESP to get it to run. A minor wrinkle.

pauljfilmalter, 2020-06-13:
Yea as chuckyro suggested also had to downgrade to 2.5.2. Working now

pauljfilmalter, 2020-06-13:
Yea as chuckyro suggested... had to downgrade to 2.5.2

gitterman, 2020-07-21:
Same problem here. Upload to ESP-01 using the ESP8266 rev. 2.7.2 fails with the same error message as  **_vigiles_**  has given.
I do use a diymore USB to serial module made for ESP-01 i. e. it has a prog/uart switch to allow easy programming.
The thing is, that it can program my four ESP-01s modules without a problem but the two ESP-01 modules, which I also have, just fail.
I did try the Arduino IDE with the  ESP8266 board rev 2.6 as well as the 2.5.2 but no success either.
It may be that both modules are broken but this thread makes me assume that it might also be some other problem.

gitterman, 2020-07-22:
SOLVED for me!
Today I got me another ESP-01 programming adapter with a CP2104 USB-Serial converter instead of the CH340 chip on the diymore.
And voilà I can upload the example programs to my ESP-01 without any problem.
So the boards are all OK.
I suppose that subtle differences in the timing easily break the communication.

Nirav-Bhadradiya, 2020-08-11:
> I didn't now that.
> Worked like a charm.
> Thanks!

hey i have similar board with me, and i am struggling to make wifi working on my board, i tried setting up dip PIN 5,6,7 but not sure what exact steps i need to perform after that, could you please guide me ? I want to send sensor data to some api from this board over wifi

pcsolns, 2020-09-26:
**MY SOLUTION**
I am using the ESP8266 12e breakout board with CH340G USB-to-Serial chip with the Arduino IDE and NodeMCU V3, but I believe this should work with any ES8266 board.

After countless wasted hours (days) trying to upload a sketch for the ESP8266, I was consistently getting the "Fatal Error Failed to connect to ESP8266" message. As an Arduino and ESP8266 newbie, it has been extremely frustrating, almost to the point of giving up with the sketch I downloaded, but here is what I discovered:
The ESP8266 can be prevented from booting if some pins are pulled low or high. So the safest way to upload a sketch is to **disconnect any external circuitry! **

A Boot failure will occur if: 
GPIO0 (D3), GPIO1 (Tx), or GPIO2 (D4) is pulled Low - ex. LED with pull-down resister to ground.
or if GPIO15 (D8) is pulled high. I stumbled on this at [https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/) 

The sketch I was trying to upload identified pins 4 and 5 to connect to LEDs so they could be controlled from a browser. The newbie mistake I made was to connect the circuitry to D4 and D5 (as identified with markings on the board) instead of pins 4 (D2) and 5 (D1). Correcting this mistake solved the problem for me. I hope my solution helps you.

OrangeGlass, 2020-09-27:
How do i get dip settings.

hellojsna, 2020-10-14:
> **disconnect any external circuitry! **

That worked for me! Thanks

imaNNeoFighT, 2020-10-14:
Here are the DIP settings.
There are 8 pins that you can configure.

[image]

shAkeN1973, 2021-01-07:
> **MY SOLUTION**
> I am using the ESP8266 12e breakout board with CH340G USB-to-Serial chip with the Arduino IDE and NodeMCU V3, but I believe this should work with any ES8266 board.
>
> After countless [...]

Thanks! Your solution helps me a lot

mitjad123, 2021-02-06:
Hi! 

Had the same problem, being lost for hours, days, surfed half of the web :-). 
My solution (Windows 10, ESP-01, USB ESP8266 ESP-01 serial programmer with switch for GPIO0 - [like this one](https://www.amazon.com/ESP-01S-Programmer-Adapter-Wireless-4-5-5-5V/dp/B07V556Q82). 
First check that pin CH_PD is high. My USB Serial Programmer did not have that, so I connected CH_PD to Vcc.
Then I made a reset Button, that connects RST pin to GND when pressed.
Now steps that work:
- I plug my USB adapter with ESP-01 on it to my PC
- I put GPIO0 pin to GND by putting switch to Prog position.
- I press my reset button, blue LED on ESP-01 blinks one
- I upload my code
- I put GPIO0 pin to "high" by putting switch to Prog position.
- I press my reset button and I can see what is going on on my Serial ....

Wow, a successful end of the day.

Aishwarya19966, 2021-03-25:
Hi,
Here i found solution https://www.youtube.com/watch?v=2iI5XK_unX4

imgonnamarryyourmom, 2021-08-15:
> go to boards manager ... find esp8266 board .. select version 2.5.2 and install it (downgrade it)

Esp downgrades always comes with another problems. It's not a surprise, why would they upgrade. I tried 2.5.0 and another error even complicated occurred

Sajad-Ghadiri, 2021-09-13:
Hope you're doing well. I had this issue on NodeMcu board with cp2102 chip. at first, I check the USB to Serial board with ch340 driver to check the correctness of NodeMcu board. it was ok, so I tried to check all wires and connections on the board from micro USB to esp8266. finally, I found the bug. there was a resistor 500 ohm between the TX port of cp2102 and RX of NodeMcu. I connect them with a wire to ignore the resistor and my code starts to upload :) . so I change the resistor to 10 ohm which is near zero. boom, everything is ok.
[image]

yarix, 2022-07-08:
Old thread, yet simple solution (once you know what you need to do).
after wasting hours i decided to re-flash the firmware and all started to work again.

🚀 Follow instructions on https://github.com/nodemcu/nodemcu-flasher to flush (use the default internal firmware)

Test code upload using:
* Arduino IDE 1.8.57.0
* ESP boards 3.0.2 (no need to use old versions like 2.5.0...)
* configure port and speed to 115200

simmani5, 2022-10-21:
I had samp problem .. and I solved just now.
나는 동일한 문제를 방금 해결하였습니다.
I did success push RST button and try upload  OK~ ^^  
리셋 버튼 한번 눌러주고 업로드 하니 잘되었습니다..

RizmyAbdulla, 2023-01-02:
i connected esp12E GPIO 0 to NODEMCU D3,it Works.....

NODEMCU ---->----- ESP-12E

GND > EN
RST                              RST
TX                                TX(IO1)
RX                                RX(IO3)
3V                                VCC > EN
GND                            GND > IO15

Hetzeilen, 2023-02-16:
Verified SOLUTION: ground the IO0 'Flash' and NEXT ground the RST pin or 'RESET'.  After this, the upload went smoothly.

mukesh1234399, 2023-03-27:
> Verified SOLUTION: ground the IO0 'Flash' and NEXT ground the RST pin or 'RESET'. After this, the upload went smoothly.

Still not working

nads232, 2023-12-02:
> Verified SOLUTION: ground the IO0 'Flash' and NEXT ground the RST pin or 'RESET'. After this, the upload went smoothly.

thanks for this it worked even though there are wires connected, also to add for others connect D3 of nodemcu to its own gnd then hold the reset button for 10 sec after releasing it try to upload the code again it worked for me

deeprajsinghsisodiya, 2023-12-31:
follow this.
while uploading reset pin of Arduino should be connected to ground. and gpoi 0 should be connected to ground.   
 esp gnd - ar gnd
esp gpio0- ar gnd
esp rx- ar rx
esp tx- ar tx
esp ch-en - ar 3.3v
esp vcc - ar 3.3v

restart ide select the board and port. 9600dau in device manager com 115200 in tools upload speed

once upload is done disconnect reset and gpio0 from ground. disconnect vcc and reconnect to 3.3v

Nhamona, 2024-01-23:
> Hi @imaNNeoFighT ,
> 
> I'm not familiar with this board but please check you have the DIP switches set for the ESP8266 to go into firmware upload mode. This board doesn't do it automatically:

> After changing the DIP switches you will need to unplug and replug the board, or press the "ESP" button which I'm guessing is a reset button for the ESP chip.

where can i go to see that DIP swicthes?

pakair, 2024-06-07:
I get the same error even when I try to upload the blink code. 

(laptop T420, Arduino IDE 1.8.19, esp-01 with FTDI connected to a usb port of the laptop)

```
esptool.py v3.0
Serial port /dev/ttyUSB3
Connecting........_____....._____....._____....._____....._____....._____.....____
A fatal esptool.py error occurred: Failed to connect to ESP8266: Timed out waiting for packet header_
```

rizmyabdulla, 2024-06-07:
> I get the same error even when I try to upload the blink code.
>
> (laptop T420, Arduino IDE 1.8.19, esp-01 with FTDI connected to a usb port of the laptop)
>
> ```
> esptool.py v3.0
> Serial port [...]

Push RST button when you upload code.

pakair, 2024-06-07:
there is no RST button either on the esp-01 or the FTDI module.

Andy8266, 2024-06-07:
When I’ve had this issue it’s been because
I’ve selected either the wrong board in the Arduino IDE or selected the
legacy boot loader instead of the new one. Some boards need their own
driver. Of course you need to be addressing the right on port.