espressif/esptool #491: esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header errore

hadimargo, 2019-12-13:
hi. i am programing esp8266 with arduino for control led by blynk.
while i uploading, this error comes:
pls help how fix this.

```

esptool.py v2.8
Serial port COM3
Connecting…………………____Traceback (most recent call last):
File “C:\Users\Webhouse\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2/tools/upload.py”, line 65, in
esptool.main(cmdline)
File “C:/Users/Webhouse/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py”, line 2890, in main
esp.connect(args.before)
File “C:/Users/Webhouse/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py”, line 483, in connect
raise FatalError(‘Failed to connect to %s: %s’ % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

```

guardineer, 2019-12-16:
same issue:
Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool\esptool.py", line 2045, in slip_reader     raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header

projectgus, 2019-12-16:
Hi @hadimargo ,

The error suggests that something is not wired up correctly so the chip isn't responding.

If you're using a different Arduino bord for the USB/serial interface, check it's wired up correctly (TXD & RXD aren't swapped) and that the chip on the Arduino board isn't also using the serial interface (if not sure, flash a blank sketch to the Arduino board's chip, first.)

Also, the 3.3V regulator on the Arduino is not rated to supply enough current for the ESP8266 so it may be that the ESP8266 is browning out.

Angus

hadimargo, 2019-12-18:
i use arduino mega and i do all u said, but it didnot work and still this errore comes. i connect rx esp8266 to tx of arduino and tx esp to rx arduino. i flash arduino also but it didnot work. pls help.

projectgus, 2019-12-18:
>  rx esp8266 to tx of arduino and tx esp to rx arduino

Pretty sure this is backwards, because you want the data to go to the USB/serial chip on the Arduino not the AVR chip.

The diagram you posted shows RX to RX and TX to TX.

underworldgamer, 2019-12-19:
Hello,

i have this one.

https://imgur.com/a/2NDum6Z

BR
Mathias

projectgus, 2019-12-19:
@underworldgamer This is a different error, please [open a new issue](https://github.com/espressif/esptool/issues/new?assignees=&labels=operation-failure&template=issue-with-attached-hardware.md&title=) and provide the details requested in the issue template.

joeservo, 2019-12-26:
does anyone know what is happening during the Connecting...___.....? is a byte being transmitted to the esp32? using an oscope i see something being sent but no response from esp32.

in a console window or using cat /dev/ttyXXXN i see
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))

waiting for download

seems like its good all is good. verified 115k baud with oscope.

pressing the reset/en button during i can see the message above being sent with oscope but esptool does not continue. 100nF cap, 10uF cap. i've tried them all.

UART0 is the correct port to be used correct? same one the message above is printed on.

grodrm00, 2020-03-28:
Hi All,

I have the same problem... I have several days trying, reading all the forums and I still have the same errors. I have check the connections, tried several Baudrates, 

I have an arduino Mega2560 board.
My WIFI-Module is ESP8266-01

The connections I had made are following this tutorial.
https://programarfacil.com/podcast/como-configurar-esp01-wifi-esp8266/

I think the problem is in the bootloader, I have read carefully this article and try all it is said
https://github.com/espressif/esptool/#entering-the-bootloader

Can anyone help me with this?
Thanks you in advance

joeservo, 2020-03-29:
what are using to communicate from PC to the ESP8266? the arduino board? if so i think that is the problem. i made my own custom board with an esp part and had a microcontroller which i used as my own USB-UART bridge and it would not communicate. i had to solder wires from a MM232R (FTDI dev board) to program the esp part. i would recommend wiring the WIFI module directly to a FTDI dev board for programming.  the MM232R i used has male pins you could wire directly to your ESP8266-1 board with single wire female jumpers. you would need a way to short reset and gpio to gnd but that could be done just by holding the pins to gnd. 

after that the next issue i had was the setting for the USB port to be used with "make flash". that was easily fixed by changing the make config settings. 

next i had to chmod 666 /dev/ttyUSB0

obviously i was using linux to program, not sure what you are using. regardless the main issue was the USB to ESP part communication method was the main issue.

grodrm00, 2020-03-29:
Thanks for your comment. I will try to find then a MM232R USB Serial UART for programming my ESP8266-01. I will tell you the results.
Thanks and regards.

chamodKanishka, 2020-04-11:
how is your results?i have the same problem

> 
> 
> Thanks for your comment. I will try to find then a MM232R USB Serial UART for programming my ESP8266-01. I will tell you the results.
> Thanks and regards.

how is your results?i have the same problem

grodrm00, 2020-04-11:
I have not buy MM232R USB Serial UART yet because of the situation I will answer you when I try this

ub3132003, 2020-07-15:
> does anyone know what is happening during the Connecting...___.....? is a byte being transmitted to the esp32? using an oscope i see something being sent but no response from esp32.
>
> in a console [...]

so am i. I was try many ESP boards. the uart has print but can't download flash

ThomHehl, 2020-11-17:
I am having this same issue:

Arduino: 1.8.13 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:
IROM   : 292608          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27728   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 2844  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 17060 ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25376 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 340240 bytes (32%) of program storage space. Maximum is 1044464 bytes.
Global variables use 45280 bytes (55%) of dynamic memory, leaving 36640 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
  File "/home/thomas/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/home/thomas/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/home/thomas/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

 I am really new to embedded software. I have followed the video and plugged a USB port from my computer to the driver board and then connected the driver board to the 7.5 inch display. I added the json file to my preferences and installed the driver. I then tried to select the board in the video, but couldn't find it. I used the NodeMCU 1.0 (ESP-12E Module) because it as 12-E. Changed my wifi connection and hit upload.

Looking at my driver board, the chip says 12-F on it. Don't know if this is an issue.

radimkarnis, 2021-03-29:
Hello, I am closing this issue due to the fact that no more information was provided by the reporter. If anyone feels like further discussion is needed or if the problem persists, please open a new issue.

closed 2021-03-29 by radimkarnis