espressif/esptool #492: finde this :http://arduino.esp8266.com/versions/2.4.1/package_esp8266com_index.json

hadimargo, 2019-12-13:
where could i find :
http://arduino.esp8266.com/versions/2.4.1/package_esp8266com_index.json

d-a-v, 2019-12-13:
https://arduino-esp8266.readthedocs.io/en/latest/installing.html
Latest version is 2.6.2, but 2.4.1 is included.

projectgus, 2019-12-16:
Hi @hadimargo,

This is the GitHub project for the esptool.py command line tool.

The ESP8266 Arduino IDE is a different project maintained by totally different people. You can find their issue tracker here:
https://github.com/esp8266/arduino/issues

However I think the problem is like @d-a-v says, this version is too old. The latest stable release package URL is https://arduino.esp8266.com/stable/package_esp8266com_index.json

closed 2019-12-16 by projectgus