espressif/esptool #493: ESP8266 flash operation failed because of a fatal error: Timed out for packet header

AHafsy, 2019-12-17:
# Operating system

# Python version: Python 2.7.6

# What Chip:  ESP8266 

# What development board or other hardware is the chip attached to

EVB: ESP-LAUNCHER

# Are you running esptool.py from an IDE such as Arduino or Eclipse?

I'm running idf.py under linux environment (Ubuntu: ESP8266_lubuntu) as it was recommended in the Get Started document.

# Full esptool.py command line that was run:

esptool.py -p /dev/ttyUSB0 flash_id

# Full output from esptool.py
**esptool.py v2.9-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: b4:e6:2d:68:3b:c4
Uploading stub...
Running stub...
Stub running...
Manufacturer: a1
Device: 4015
Detected flash size: 2MB
Hard resetting via RTS pin...**

I should turn off and then turn on the EVB every time to get those information. Otherwise, there is no output.

# The problem:
My problem is when I want to flash data on the board I get : A fatal error occurred: Timed out waiting for packet header. 
After running "idf.py menuconfig", and make the wanted configuration I run "idf.py -p /dev/ttyUSB0 flash" and I get this error:

**Executing action: flash
Running esptool.py in directory /home/esp8266/hello_world/build
Executing "/home/esp8266/.espressif/python_env/idf4.1_py2.7_env/bin/python /home/esp8266/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash @flash_project_args"...
esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size 16MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v2.9-dev
Serial port /dev/ttyUSB0
Connecting....

A fatal error occurred: Timed out waiting for packet header
esptool.py failed with exit code 2**

Could you please help me to fix this problem and flash the data on the EVB ?

projectgus, 2019-12-17:
Hi @AHafsy,

Thanks for the detailed description.

The ESP-LAUNCHER has no automatic circuit to reset it into a bootloader mode each time it needs flashing. So you need to set the IO0 Control switch to low, and then toggle the 5V Power switch off and then on again each time before running esptool.py (it's also possible to do this quickly while esptool.py is in the "Connecting..." stage).

A diagram showing which switches to set can be found in the ESP8266 Quick Start guide section "1.3. Downloading Firmware into the ESP-LAUNCHER":
https://www.espressif.com/sites/default/files/documentation/esp8266_quick_start_guide_en.pdf

(The steps in the Guide are for using the GUI flasher tool, but you can use esptool.py instead - just configure the switches as described, and when it says to press the START button this is the time to run esptool.py or the flashing step in the SDK.)

AHafsy, 2019-12-19:
Hello @projectgus,
Thank you, it helped me.
I choose to flash it using the GUI flasher which easier and quicker and it works. 
Also, I followed the steps described on the document you mentioned and I have a new problem. Could you please help to figure out the source of this problem ? 
I'm opening a new issue for this: https://github.com/espressif/ESP8266_RTOS_SDK/issues/786

projectgus, 2019-12-19:
@AHafsy Glad you got flashing working correctly.

I don't know anything about the ESP8266 RTOS SDK, sorry. But someone from that project should be able to help.

closed 2019-12-19 by projectgus