espressif/esptool #494: Getting error "A fatal error occurred: MD5 of file does not match data in flash!" after uploading flash ESP WROOM 02

EdderHdez, 2019-12-18:
I keep getting this error after flash upload of custom build

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 28156, room 16 
tail 12
chksum 0x9b
ho 0 tail 12 room 4
load 0x3ffe8000, len 2568, room 12 
tail 12
chksum 0x98
ho 0 tail 12 room 4
load 0x3ffe8a08, len 8, room 12 
tail 8
chksum 0xd5
csum 0xd5
system SPI FI size:3, Flash size: 2097152
boot not set
ota1 not set
ota2 not set
rf cal sector: 11
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 3.0.1-dev(fce080e) compiled @ Apr 16 2019 15:02:05
phy ver: 1145_0, pp ver: 10.2

Microcontroller | ESP8266
Frequency | 80MHz
Flash | 2MB
RAM | 80KB

I've been reading a lot but still have not found the solution

After erease flash with `python esptool.py -p COM15 erase_flash` i get (baud - 74880) this message:

ets Jan  8 2013,rst cause:1, boot mode:(3,7)
ets_main.c

projectgus, 2019-12-18:
Hi @EdderHdez ,

If you got the error mentioned in the title (MD5 of file does not match data in flash) then it's expected thart the ESP8266 will not boot successfully, because the flash contents are not connect.

There isn't enough information in your post to make any other diagnosis.

Please [answer all the questions in the issue template](https://github.com/espressif/esptool/blob/master/.github/ISSUE_TEMPLATE/issue-with-attached-hardware.md#what-development-board-or-other-hardware-is-the-chip-attached-to). In particular, the details about hardware and connections and the full esptool.py line and the full output from esptool.py when it fails.

These questions are there so that we can help you debug the problem.

radimkarnis, 2020-12-09:
Hello,
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-09 by radimkarnis