espressif/esptool #495: Windows NODEMCU OK but Linux Esptool : Failed to connect to Espressif device: Timed out waiting for packet header

sebastienPoussard, 2019-12-24:
Dear projectgus, 

* Operating system : Debian buster 10
* Python 2.7.16  &  Python 3.7.3
* Chip : ESP8266 with Atmega32U4 USB chip [AliExpress link](https://aliexpress.com/item/32946940424.html)
* esptool is lauched directly from terminal 
```Bash 
esptool --baud 9600 --trace --port /dev/ttyACM0 write_flash 0x00000 fichiers/esp8266_wifi_duck_4mb.bin
```
* output : 
```Bash
[..... same data ....]
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
_TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```
* additional informations
  * to enter into flashing mode, i have to solder 2 point on the chip which i did.
  * flashing the chip on Windows work with NODEMCU but i would like to troubleshot the issue on Linux
  * followed your troubleshoot guide without any success, i launched minicom but no output on the screen

* ModeManager is descativated, flashing the Atmega32U4 with Arduino IDE works.

Best regards.

joeservo, 2019-12-27:
timeout 0.100
if thats true why? why not write a 1 second timeout or greater. 

i'm getting the same results. decided to buy every different dev kit i could find to see which will work.

sebastienPoussard, 2019-12-31:
> timeout 0.100
> if thats true why? why not write a 1 second timeout or greater.
> 
> i'm getting the same results. decided to buy every different dev kit i could find to see which will work.

Dear Joeservo, 
I don't think (at least in my case) that the problem is linked to 0,1 sec timeout,
I suspect something releveant to Linux OS such as ModemManager when you try to flash with Arduino IDE.

joeservo, 2019-12-31:
you pass -baud 9600 in your arguments. why so? the ESP32 communicates at 115200.

does this beetle dev board you provided a link for have a boot and/or reset switch or is it done with the RTS/DTR pins. do you have a schematic for it?

sebastienPoussard, 2020-01-04:
> you pass -baud 9600 in your arguments. why so? the ESP32 communicates at 115200.
> 
> does this beetle dev board you provided a link for have a boot and/or reset switch or is it done with the RTS/DTR pins. do you have a schematic for it?

Dear joeservo,

I initially tried without the baud argument it did not worked as well. I tried with the baud at 9600 as specified in the troubleshoot guide to eliminate a potential stability error.
As previously specified, the board is a ESP8266 not ESP32.

I will try again when possible though and will report my result.

Best regards.

radimkarnis, 2020-12-11:
Hello @sebastienPoussard,

is there any update? have you been able to resolve your problem? Please let me know

Radim

radimkarnis, 2020-12-18:
Hello everyone,
I am closing this issue because it seems like it is not relevant anymore. Please reopen if you think further discussion is needed.

closed 2020-12-18 by radimkarnis