espressif/esptool #496: Regression; Can not detect flash size after v2.6

awigen, 2019-12-31:
The chip in question is a **ESP-WROOM-02**.

The correct output looks like:

```
$ ./esptool.py flash_id
esptool.py v2.5.1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:c6:3a:df:dd:5d
Uploading stub...
Running stub...
Stub running...
Manufacturer: 0e
Device: 4015
Detected flash size: 2MB
Hard resetting via RTS pin...
```

However this is broken in v2.6:
```

$ ./esptool.py flash_id
esptool.py v2.6
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:c6:3a:df:dd:5d
Uploading stub...
Running stub...
Stub running...
Manufacturer: 00
Device: 0000
Detected flash size: Unknown
Hard resetting via RTS pin...
```

git bisect reveals this commit:

```
302c7bf2475805a67ec08a537897f7555e62d52d is the first bad commit
commit 302c7bf2475805a67ec08a537897f7555e62d52d
Author: Angus Gratton <angus@espressif.com>
Date:   Fri Dec 21 12:20:35 2018 +1100

    esp8266: Call spi_flash_attach() when stub runs
    
    Necessary if the ESP8266 has been woken from deep sleep,
    when SPI flash is still powered down.
    
    Closes https://github.com/espressif/esptool/issues/351
    
    Probably also the fix for https://github.com/espressif/esptool/issues/380

 esptool.py                  | 134 ++++++++++++++++++++++----------------------
 flasher_stub/stub_flasher.c |   2 +
 test/test_esptool.py        |  22 ++++++++
 3 files changed, 91 insertions(+), 67 deletions(-)
```

projectgus, 2020-02-10:
Hi @awigen,

Thanks for reporting this and sorry for the slow response.

Do other esptool functions (apart from flash size detection) work on the more recent versions?

radimkarnis, 2020-12-11:
Hello @awigen,

is there any update? have you been able to verify other esptool (>v2.6) functions working with ESP-WROOM-02? 

Radim

radimkarnis, 2020-12-18:
I am closing this issue because it seems like it is not relevant anymore. Please reopen if you think further discussion is needed.

closed 2020-12-18 by radimkarnis