espressif/esptool #497: Documentation missing

howroyd, 2020-01-02:
This is more of a request for documentation or a statement of the replies sent by the ROM bootloader (etc) to certain commands.

In my case, when sending the first block of a new app binary, the reply I get is:
`C0 01 03 04 00 07 12 20 07 01 07 00 00 C0`

The status `0x07` tells me a checksum error but what is the data in the value/checksum field of this message?  I am assuming it isn't the checksum the ROM bootloader calculated, since this should be an 8 bit value in a little endian 32 bit word.

Do the four bytes `07 12 20 07` mean anything???

projectgus, 2020-02-10:
Hi @howroyd ,

Sorry again for slow reply.

Have you seen the description here?
https://github.com/espressif/esptool/wiki/Serial-Protocol#checksum

The checksum for these data commands is stored in the 8-bit checksum header field:
https://github.com/espressif/esptool/wiki/Serial-Protocol#command
(which is ignored for other commands)

> Do the four bytes 07 12 20 07 mean anything???

No, for all commands except READ_REG this is junk data.

howroyd, 2020-02-13:
Many thanks, I'll close this now.

Cheers

closed 2020-02-13 by howroyd