espressif/esptool #498: FLASH_END command not being sent - is it not required?

howroyd, 2020-01-03:
* Operating system: Win10
* Python version: (`python -V` to check this)
* esptool.py version: v2.8

# Full esptool.py command line as run:

`esptool.py --trace --no-stub --port COM10 -b 115200 write_flash 0x1000 build\blink.bin -u`

# Full output from esptool.py

```
TRACE +0.004 Write 1050 bytes:
    c000031004ca0000 00000400008f0000 | ................
    0000000000000000 003641001df00000 | .........6A.....
    0000000000000000 ce7020e0a22eeee5 | .........p .....
    e2e56e2600abefb7 ce77f00af02bacb3 | ..n&.....w...+..
    ebbaeb863bc7bb37 31ffffffffffffff | ....;..71.......
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffffffffffffffff | ................
    ffffffffffffffff ffc0             | ..........
TRACE +0.104 Read 1 bytes: c0
TRACE +0.001 Read 13 bytes: 010304002040160000000000c0
TRACE +0.001 Received full packet: 010304002040160000000000
Wrote 147456 bytes at 0x00001000 in 16.9 seconds (69.7 kbit/s)...
TRACE +0.004 command op=0x13 data len=16 wait_response=1 timeout=3.000 data=00100000303c02000000000000000000
TRACE +0.000 Write 26 bytes:
    c000131000000000 0000100000303c02 | .............0<.
    0000000000000000 00c0             | ..........
TRACE +0.213 Read 1 bytes: c0
TRACE +0.002 Read 45 bytes:
    0113240020401600 6464393562373962 | ..$. @..dd95b79b
    6662363238656135 3830613965636562 | fb628ea580a9eceb
    3538353536643963 00000000c0       | 58556d9c.....
TRACE +0.002 Received full packet:
    0113240020401600 6464393562373962 | ..$. @..dd95b79b
    6662363238656135 3830613965636562 | fb628ea580a9eceb
    3538353536643963 00000000         | 58556d9c....
Hash of data verified.

Leaving...
Hard resetting via RTS pin..
```.

# What is the expected behaviour?

Send command `0x04 FLASH_END`.

# What actually happened?

As you can see from the trace, no FLASH_END message was sent, instead it goes straight to reading the MD5.

# Question

Am I correct in therefore assuming that the FLASH_END command is superfluous and not required?  I'm scratching my head over the documentation.

# Reason

I am writing a cpp app that will ultimately go on a microcontroller to flash an ESP32 using the ROM bootloader.

projectgus, 2020-02-10:
Hi @howroyd ,

Sorry for the slow reply.

`FLASH_END` is unfortunately named because it doesn't really go with `FLASH_BEGIN`/`FLASH_DATA` , what it actually does is exits the UART bootloader (and optionally soft-resets the chip or runs the "user code").

So it is used, but as part of the soft reset cycle for ESP8266:
https://github.com/espressif/esptool/blob/939b3e9/esptool.py#L1012

ESP32 unfortunately can't use this command reliably at all, so we always do a hard reset or stay in bootloader.

Bottom line is that if you're OK to stay in UART downloader mode (or you plan to trigger a hard reset) then you can skip sending it, no ill effects.

By the way, two weeks ago we released an open source cross-platform library for flashing an ESP32 from another MCU. May be of interest: https://github.com/espressif/esp-serial-flasher/

Angus

closed 2020-02-10 by projectgus

projectgus, 2020-02-10:
Hi @howroyd ,

I just noticed that the Serial Protocol wiki page had a vague and incomplete description of `FLASH_END`, so I fixed it. Sorry about that.

Angus